SeisComP3/seiscomp3 #163: Potential problem with seiscomp3 fdsnws module

rannof, 2018-02-19:
Hi,
I'm not sure if its a feature or bug nor if it is a fixable issue with obspy or seiscomp3.
When using obspy to download stream data from an fdsnws server spawned by seiscomp3 (seiscomp3-jakarta-2017.334.01-centos7-x86_64),
the server is hanged with CLOSE_WAIT on the ports. After several reads, the server reaches its client limit and freeze. seiscomp status shows it works OK.
============== example of client side ================

```
from obspy.clients.fdsn import Client as fdsnclient
.
.
.
def get_data(self):
    for tr_id in tr_id_list: # loop over several trace ids
        self.flink = fdsnclient(self.fdsnserver, timeout=30)
        stream = self.flink.get_waveforms(tr_id.network, tr_id.station, tr_id.location, tr_id.channel, self.starttime, self.endtime)
    .
    .
    .
(exit python script, re-run it after a while)
```
==============================================

=========== output on server ============
```

> netstat -a | grep 8181
tcp       27      0 0.0.0.0:8181            0.0.0.0:*               LISTEN     
tcp      230      0 ws-fdsn:8181            clientip:54342 CLOSE_WAIT 
tcp      611      0 ws-fdsn:8181             clientip:58324 CLOSE_WAIT
.
.
.
(similar entries as the limit of clients on the module)
```
====================================

A possible workaround (not sure it really solves the problem):
============== No problem code ==========

```
def get_data(self):
    from obspy.clients.fdsn import Client as fdsnclient # import within the function)
    for tr_id in tr_id_list: # loop over several trace ids
        self.flink = fdsnclient(self.fdsnserver, timeout=30)
        stream = self.flink.get_waveforms(tr_id.network, tr_id.station, tr_id.location, tr_id.channel, self.starttime, self.endtime)
    del self.flink # delete the client object
```
=====================================

ObsPy client version, Python version and Platform (Windows, OSX, Linux ...)
obspy v 1.0.2 (possibly also older and newer versions), osx, linux (windows not tested)
How did you install ObsPy and Python (pip, anaconda, from source ...)
pip, anaconda
If possible please supply a short, self contained, correct example that
demonstrates the issue.
If this is a regression (used to work in an earlier version of ObsPy),
please note when it used to work.
Do not know.
similar issue was added to obspy github page: #2078.

gempa-jabe, 2018-02-19:
What record source have you configured with your fdsnws?

rannof, 2018-02-19:
global.cfg:
# Name of the recordstream service implementation.
recordstream.service = combined
# Service specific parameters like a IP address or a file name to use.
recordstream.source = slink/127.0.0.1:18000;sdsarchive/[PATHTOARCHIVE]
fdsnws.cfg:
# localhost.
listenAddress = 0.0.0.0
# Number of maximum simultaneous requests.
connections = 200
recordBulkSize = 1024

gempa-jabe, 2018-02-19:
You should never use a realtime stream with fdsnws. The Seedlink connection will block until data for requested channels are available. That does not play well with fdsnws. Use exclusively sdsarchive or slink with options ```?timeout=1&retries=0```.

rannof, 2018-02-19:
I'll try that.
Thanks!

gempa-jabe, 2018-02-19:
I would like to ask you to post non developer questions in the forums at https://forum.seiscomp3.org next time. Thank you.

gempa-jabe, 2018-02-19:
Related to obspy/obspy#2078.

rannof, 2018-02-19:
Sure, I thought it is development related.

Ran Novitsky Nof (PhD)

Seismologist

Seismological Division

The Geological Survey of Israel

Web: http://ran.rnof.info/

petrrr, 2018-03-01:
Are there any news on this issue?

We at our datacenter observe a serious service stability problem with fdsnws dataselect. The problem appeared after the update of Seiscomp3 to version 2017.334.03. Since then the service hangs regularly, creating a disservice, requiring a restart of the service (only FDSN).

Our problem seems to be exactly the one reported here.

gempa-jabe, 2018-03-01:
The answer has been posted above. In short: don't use blocking streaming sources such as Seedlink for FDSNWS.

rannof, 2018-03-01:
I can confirm that after adding the retry and timeout parameters to the
seedlink server settings, no hangouts were reported.

Ran Novitsky Nof (PhD)

Seismologist

Seismological Division

The Geological Survey of Israel

Web: http://ran.rnof.info/

petrrr, 2018-03-01:
I was quite sure we have no blocking streams configured and I rechecked. Most of the stuff below is commented, the active one is a normal Posix filesystem.

```
#recordstream.service = sdsarchive
#recordstream.source = /home/sysop/seiscomp3/var/lib/archive

# ok per verce
recordstream.service = sdsarchive
recordstream.source = /mnt/seedstore_nfs/archive

#recordstream.service = slink
#recordstream.source = //

#recordstream.service = combined
#recordstream.source = slink//;sdsarchive//mnt/seedstore_nfs/archive

#recordstream.source = sdsarchive//mnt/seedstore_nfs/archive;sdsarchive//home/sysop/seiscomp3/var/lib/archive??rtMax=1800
```

petrrr, 2018-03-01:
Looks like a corrupt file might cause the problem, however the archive has not changed with the update and the service was resilient before.

```
cat /home/sysop/.seiscomp3/log/fdsnws.log 
2018/03/01 08:04:37 [debug/SDSARCHIVE] SDS request: 2016,08,31,21,54,40 2016,09,01,03,04,41 IV BDI BHE
2018/03/01 08:04:37 [debug/SDSARCHIVE] + /mnt/archive_aufs/2016/IV/BDI/BHE.D/IV.BDI..BHE.D.2016.244 (init:1)
2018/03/01 08:04:37 [debug/SDSARCHIVE] + /mnt/archive_aufs/2016/IV/BDI/BHE.D/IV.BDI..BHE.D.2016.245 (not found)
2018/03/01 08:04:37 [debug/SDSARCHIVE] SDS request: 2016,08,31,21,54,40 2016,09,01,03,04,41 IV BDI BHE
2018/03/01 08:04:37 [debug/SDSARCHIVE] + /mnt/archive_aufs/2016/IV/BDI/BHE.D/IV.BDI..BHE.D.2016.244 (init:1)
2018/03/01 08:04:37 [debug/SDSARCHIVE] + /mnt/archive_aufs/2016/IV/BDI/BHE.D/IV.BDI..BHE.D.2016.245 (not found)
2018/03/01 08:04:37 [debug/SDSARCHIVE] SDS request: 2016,08,31,21,54,40 2016,09,01,03,04,41 IV BDI BHE
2018/03/01 08:04:37 [debug/SDSARCHIVE] + /mnt/archive_aufs/2016/IV/BDI/BHE.D/IV.BDI..BHE.D.2016.244 (init:1)
2018/03/01 08:04:37 [debug/SDSARCHIVE] + /mnt/archive_aufs/2016/IV/BDI/BHE.D/IV.BDI..BHE.D.2016.245 (not found)
2018/03/01 08:04:37 [debug/SDSARCHIVE] SDS request: 2016,08,31,21,54,40 2016,09,01,03,04,41 IV BDI BHE
```

gempa-jabe, 2018-03-01:
Please check the CPU load and if possible, upload the corrupt file for further inspection.

petrrr, 2018-03-01:
Here the file:
 * [IV.BDI..BHE.D.2016.244.zip](https://github.com/SeisComP3/seiscomp3/files/1770624/IV.BDI.BHE.D.2016.244.zip)

The hypothesis is that the error handling might cause some infinite loop. But we have no clue which piece of code this corresponds to.

gempa-jabe, 2018-03-01:
I have an idea. Can you check the CPU load? I suspect that fdsnws runs at 100%.

gempa-jabe, 2018-03-01:
I can confirm the issue and am working on a fix.

gempa-jabe, 2018-03-01:
I have committed a patch: d6d19a435c9ab04a49e728d20f0c141d0bcac851. You can apply it to the current release source (it maintains the ABI) and give it a try. I have successfully tested it with your file.

vlauciani, 2018-03-02:
Dear all

I recompiled the last version `2017.334.04`, using the patch on `./src/trunk/libs/seiscomp3/io/recordstream/sdsarchive.cpp` file; to test in production server which binary I have to substitute?

gempa-jabe, 2018-03-02:
You have to substitute ```libseiscomp3_core.so``` and restart fdsnws.

vlauciani, 2018-03-03:
Hi Jan

When I substitute the library `libseiscomp3_core.so`, running `seiscomp stop` and `seiscomp start` I receive this **error** message:
```
$ seiscomp stop
error: /home/sysop/seiscomp3/etc/init/arclinkproxy.py: No module named _System
error: /home/sysop/seiscomp3/etc/init/trunk.py: No module named _System
error: /home/sysop/seiscomp3/etc/init/arclink.py: No module named _Core
slmon is not running
shutting down slarchive
shutting down seedlink
scwfparam is not running
scvsmaglog is not running
scvsmag is not running
scvoice is not running
scsohlog is not running
screloc is not running
scqc is not running
scproclat is not running
scmag is not running
scm is not running
scimport is not running
scimex is not running
scevtlog is not running
scevent is not running
scenvelope is not running
scdb is not running
scautopick is not running
scautoloc is not running
scamp is not running
scalert is not running
ql2sc is not running
fdsnws is not running
ew2sc3 is not running
scmaster is not running
shutting down spread
$
```

jsaul, 2018-03-03:
> When I substitute the library libseiscomp3_core.so, running seiscomp stop and seiscomp start I receive this error message:

Which SC3 version are you using?

vlauciani, 2018-03-03:
> Which SC3 version are you using?

the `2017.334.03`. Have I to use the `2017.334.04`?

gempa-jabe, 2018-03-03:
You can check with ```ldd``` _System.so and _Core.so (in seiscomp3/lib/python/seiscomp3) if libseiscomp3_core.so can be found. Both patch releases are binary compatible, at least they are supposed to be.

You can also check the output of e.g.

```
seiscomp exec scsohlog --debug
```

or any other Python app. It should print why it fails. I have currently not concrete idea.

vlauciani, 2018-03-04:
This is the output, after the patch is applied:

- Running the `scsohlog`:
```
sysop@eida:~$
sysop@eida:~$
sysop@eida:~$ seiscomp exec scsohlog --debug
error: /home/sysop/seiscomp3/etc/init/arclinkproxy.py: No module named _System
error: /home/sysop/seiscomp3/etc/init/trunk.py: No module named _System
error: /home/sysop/seiscomp3/etc/init/arclink.py: No module named _Core
Traceback (most recent call last):
  File "/home/sysop/seiscomp3/bin/scsohlog", line 15, in <module>
    import sys, os, time, re, seiscomp3.Client, seiscomp3.System
  File "/home/sysop/seiscomp3/lib/python/seiscomp3/Client.py", line 17, in <module>
    _Client = swig_import_helper()
  File "/home/sysop/seiscomp3/lib/python/seiscomp3/Client.py", line 16, in swig_import_helper
    return importlib.import_module('_Client')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named _Client
sysop@eida:~$
sysop@eida:~$
```

- Running `ldd`:
```
sysop@eida:~/seiscomp3/lib/python/seiscomp3$
sysop@eida:~/seiscomp3/lib/python/seiscomp3$
sysop@eida:~/seiscomp3/lib/python/seiscomp3$ ldd _Core.so
./_Core.so: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.14' not found (required by /home/sysop/seiscomp3/lib/libseiscomp3_core.so)
./_Core.so: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `CXXABI_1.3.8' not found (required by /home/sysop/seiscomp3/lib/libseiscomp3_core.so)
./_Core.so: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by /home/sysop/seiscomp3/lib/libseiscomp3_core.so)
./_Core.so: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /home/sysop/seiscomp3/lib/libseiscomp3_core.so)
	linux-vdso.so.1 =>  (0x00007ffc4fbfc000)
	libpython2.7.so.1.0 => /usr/lib/libpython2.7.so.1.0 (0x00007f22eee74000)
	libseiscomp3_core.so => /home/sysop/seiscomp3/lib/libseiscomp3_core.so (0x00007f22ed35e000)
	libboost_thread.so.1.49.0 => /usr/lib/libboost_thread.so.1.49.0 (0x00007f22ef8a8000)
	libboost_iostreams.so.1.49.0 => /usr/lib/libboost_iostreams.so.1.49.0 (0x00007f22ef88d000)
	libboost_filesystem.so.1.49.0 => /usr/lib/libboost_filesystem.so.1.49.0 (0x00007f22ef86c000)
	libboost_regex.so.1.49.0 => /usr/lib/libboost_regex.so.1.49.0 (0x00007f22ef742000)
	libboost_program_options.so.1.49.0 => /usr/lib/libboost_program_options.so.1.49.0 (0x00007f22ef6dc000)
	libboost_system.so.1.49.0 => /usr/lib/libboost_system.so.1.49.0 (0x00007f22ef6d8000)
	libxml2.so.2 => /usr/lib/x86_64-linux-gnu/libxml2.so.2 (0x00007f22ecffe000)
	libssl.so.1.0.0 => /usr/lib/x86_64-linux-gnu/libssl.so.1.0.0 (0x00007f22ecd9d000)
	libcrypto.so.1.0.0 => /usr/lib/x86_64-linux-gnu/libcrypto.so.1.0.0 (0x00007f22ec9a3000)
	libbson-1.0.so.1 => /home/sysop/seiscomp3/lib/libbson-1.0.so.1 (0x00007f22ec760000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f22ec558000)
	libseiscomp3_config.so => /home/sysop/seiscomp3/lib/libseiscomp3_config.so (0x00007f22ec310000)
	libmseed.so.2.17 => /home/sysop/seiscomp3/lib/libmseed.so.2.17 (0x00007f22ec0e4000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f22ebddd000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f22ebb5b000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f22eb945000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f22eb5b8000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f22eb3a1000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f22eb185000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f22eaf81000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f22ead7e000)
	libboost_thread.so.1.62.0 => not found
	libboost_iostreams.so.1.62.0 => not found
	libboost_filesystem.so.1.62.0 => not found
	libboost_regex.so.1.62.0 => not found
	libboost_program_options.so.1.62.0 => not found
	libboost_system.so.1.62.0 => not found
	libssl.so.1.1 => not found
	libcrypto.so.1.1 => not found
	libbz2.so.1.0 => /lib/x86_64-linux-gnu/libbz2.so.1.0 (0x00007f22eab6b000)
	libicuuc.so.48 => /usr/lib/x86_64-linux-gnu/libicuuc.so.48 (0x00007f22ea7b3000)
	libicui18n.so.48 => /usr/lib/x86_64-linux-gnu/libicui18n.so.48 (0x00007f22ea386000)
	libicudata.so.48 => /usr/lib/x86_64-linux-gnu/libicudata.so.48 (0x00007f22e9016000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f22e8df2000)
[... 26 lines cut ...]
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f1f182fa000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f1f18078000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f1f17e62000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f1f17ad5000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f1f178be000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f1f176a2000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f1f1749e000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f1f1729b000)
	libboost_thread.so.1.62.0 => not found
	libboost_iostreams.so.1.62.0 => not found
	libboost_filesystem.so.1.62.0 => not found
	libboost_regex.so.1.62.0 => not found
	libboost_program_options.so.1.62.0 => not found
	libboost_system.so.1.62.0 => not found
	libssl.so.1.1 => not found
	libcrypto.so.1.1 => not found
	libbz2.so.1.0 => /lib/x86_64-linux-gnu/libbz2.so.1.0 (0x00007f1f17088000)
	libicuuc.so.48 => /usr/lib/x86_64-linux-gnu/libicuuc.so.48 (0x00007f1f16cd0000)
	libicui18n.so.48 => /usr/lib/x86_64-linux-gnu/libicui18n.so.48 (0x00007f1f168a3000)
	libicudata.so.48 => /usr/lib/x86_64-linux-gnu/libicudata.so.48 (0x00007f1f15533000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f1f1530f000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f1f1bc71000)
sysop@eida:~/seiscomp3/lib/python/seiscomp3$
sysop@eida:~/seiscomp3/lib/python/seiscomp3$
```

gempa-jabe, 2018-03-05:
Are you sure that you compiled the lib for the correct OS? It looks like a binary issue with your installation and the library you copied over.

vlauciani, 2018-03-06:
> Are you sure that you compiled the lib for the correct OS? It looks like a binary issue with your installation and the library you copied over.

Hi Jan,
thank you for you feedback, I found the compilation problem.
Now seems to be ok and the library is in production; I'll test it for a few days and then I let you know.

vlauciani, 2018-03-08:
Hi Jan

The **dataselect** service works perfectly; thank you for the patch.
I think you can close the issue.

Valentino

gempa-jabe, 2018-03-08:
I am glad to hear that. Thanks for the feedback.

petrrr, 2018-03-14:
@gempa-jabe: We again observe regular hangs. The file is now a different one, but the behavior seems to be rather similar. It looks like some routine enters an infinite loop.

Here the output of the log:
```
cat /home/sysop/.seiscomp3/log/fdsnws.log
[...]
2018/03/14 08:46:39 [debug/SDSARCHIVE] SDS request: 2010,01,06,16,48,30 2010,01,06,16,54,31 IV ARVD BHE 
2018/03/14 08:46:39 [debug/SDSARCHIVE] + /mnt/archive_aufs/2010/IV/ARVD/BHE.D/IV.ARVD..BHE.D.2010.006 (init:1)
2018/03/14 08:46:39 [debug/SDSARCHIVE] SDS request: 2010,01,06,16,48,30 2010,01,06,16,54,31 IV ARVD BHE 
2018/03/14 08:46:39 [debug/SDSARCHIVE] + /mnt/archive_aufs/2010/IV/ARVD/BHE.D/IV.ARVD..BHE.D.2010.006 (init:1)
2018/03/14 08:46:39 [debug/SDSARCHIVE] SDS request: 2010,01,06,16,48,30 2010,01,06,16,54,31 IV ARVD BHE 
2018/03/14 08:46:39 [debug/SDSARCHIVE] + /mnt/archive_aufs/2010/IV/ARVD/BHE.D/IV.ARVD..BHE.D.2010.006 (init:1)
2018/03/14 08:46:39 [debug/SDSARCHIVE] SDS request: 2010,01,06,16,48,30 2010,01,06,16,54,31 IV ARVD BHE 
2018/03/14 08:46:39 [debug/SDSARCHIVE] + /mnt/archive_aufs/2010/IV/ARVD/BHE.D/IV.ARVD..BHE.D.2010.006 (init:1)
[...]
```

The referenced file seems to be corrupt:
```
sysop@eida:~$ qmerge -n /mnt/archive_aufs/2010/IV/ARVD/BHE.D/IV.ARVD..BHE.D.2010.006
partial block of 163 bytes at end of /mnt/archive_aufs/2010/IV/ARVD/BHE.D/IV.ARVD..BHE.D.2010.006
ARVD.IV.BHE. rate=20 (2010.006 00:00:00.8450 to 2010.006 00:13:24.1452) : 16066 points, 0.2 msec correction, (min,max,max_step = 0.0,0.3,0.2 msec)
```

Should we reopen this issue or open a new one?

gempa-jabe, 2018-03-14:
I can have a look if you attach the file.

petrrr, 2018-03-14:
Sure! I just waited to understand if we continue here.

Here the file, causing the new problem.

[IV.ARVD..BHE.D.2010.006.zip](https://github.com/SeisComP3/seiscomp3/files/1810856/IV.ARVD.BHE.D.2010.006.zip)

gempa-jabe, 2018-03-14:
Works for me. Are you sure that you have applied the patch?

closed 2018-03-14 by gempa-jabe

gempa-jabe, 2018-03-14:
I closed the issue because the behaviour you described was identical to the initial issue. If you still observe hangs and if you can confirm that you applied the fix, then please reopen the issue.

petrrr, 2018-03-14:
@gempa-jabe: According to internal communication of my colleague @vlauciani who took are of this the patch was applied. I'll check back with him and try to understand better and we let you know.

vlauciani, 2018-03-22:
Hi @gempa-jabe 

The patch was applied and the "original" hang was solved; now we have an other error that hang the service; the `~/.seiscomp3/log/fdsnws.log` is completely full with lines like this:
```
. . .
2018/03/22 07:48:47 [error/SDSARCHIVE] file read exception: Invalid miniSEED header
2018/03/22 07:48:47 [error/SDSARCHIVE] file read exception: Invalid miniSEED header
2018/03/22 07:48:47 [error/SDSARCHIVE] file read exception: Invalid miniSEED header
2018/03/22 07:48:47 [error/SDSARCHIVE] file read exception: Invalid miniSEED header
2018/03/22 07:48:47 [error/SDSARCHIVE] file read exception: Invalid miniSEED header
2018/03/22 07:48:47 [error/SDSARCHIVE] file read exception: Invalid miniSEED header
2018/03/22 07:48:47 [error/SDSARCHIVE] file read exception: Invalid miniSEED header
2018/03/22 07:48:47 [error/SDSARCHIVE] file read exception: Invalid miniSEED header
2018/03/22 07:48:47 [error/SDSARCHIVE] file read exception: Invalid miniSEED header
. . .
```

I cannot understand which MSEED (or request) generate this error beacuse all logs availabe are full of these lines:
```
$ ll ~/.seiscomp3/log/fdsnws*
-rw-r--r-- 1 sysop sysop  21020503 Mar 22 08:00 /home/sysop/.seiscomp3/log/fdsnws.log
-rw-r--r-- 1 sysop sysop 104857620 Mar 22 07:48 /home/sysop/.seiscomp3/log/fdsnws.log.1
-rw-r--r-- 1 sysop sysop 104857620 Mar 22 07:48 /home/sysop/.seiscomp3/log/fdsnws.log.2
-rw-r--r-- 1 sysop sysop 104857620 Mar 22 07:47 /home/sysop/.seiscomp3/log/fdsnws.log.3
-rw-r--r-- 1 sysop sysop 104857620 Mar 22 07:47 /home/sysop/.seiscomp3/log/fdsnws.log.4
-rw-r--r-- 1 sysop sysop 104857620 Mar 22 07:46 /home/sysop/.seiscomp3/log/fdsnws.log.5
-rw-r--r-- 1 sysop sysop 104857620 Mar 22 07:46 /home/sysop/.seiscomp3/log/fdsnws.log.6
-rw-r--r-- 1 sysop sysop 104857620 Mar 22 07:45 /home/sysop/.seiscomp3/log/fdsnws.log.7
$
```

Any idea?

gempa-jabe, 2018-03-22:
> Any idea?

I don't have an idea which file on your side produces such an output. There should be a debug output before the error indicating the file which is currently read. Can you identify it or did you turn off debug logging?

petrrr, 2018-03-22:
@gempa-jabe: 
Indeed as @vlauciani wrote above our problem is that we lost the information about which file exactly causes the problem, so we cannot inspect or test. And this already happened before. However, we do **not** loose this information because the debug level logging was turned off. 

Instead the information (I think there are few lines with the request, streams, etc. in the log) is simple flushed out of the logs. The related error message  (`[error/SDSARCHIVE] file read exception: Invalid miniSEED header`) is produced at such a high rate, that all available logfiles were filled within few minutes. Just observe the modification time of the files above. And for comparison: The problem was first detected yesterday night at 21:55:37 UTC.

So what we could do to conserve the information without then blocking the server completely?

We'll also try to recover some useful information independently from SC3, but we are not sure that is possible.

BTW: I think we are now dealing with a slightly different problem. Should we move to a separate issue?

gempa-jabe, 2018-03-22:
I also think that this is a different issue. This issue dealt with an infinite loop which was fixed. Now we are dealing with either corrupt files or a bad file read index. Please open a new issue for that.