SeisComP3/seiscomp3 #170: What's the version of Python and Perl that seiscomp3 supports?

CovMat, 2018-04-27:
Dear Developer:

I am using python 3.6.4 and perl 5.16.3 in CentOS 7 x86_64:
When I try to compile or run seiscomp3, some errors occur.  This is the result when I ```make -f Makefile.cvs``` and press ```c``` to configure:
```
 CMake Error at src/system/libs/python/CMakeLists.txt:1 (IF):
   if given arguments:

     "File \"<string>\", line 1
       import sys" "print sys.version[:3]
                          ^

   SyntaxError: invalid syntax" "STREQUAL" "2.3"

   Unknown arguments specified

Errors occurred during the last pass                                            
                                                          CMake Version 2.8.12.2
Press [e] to exit help

```

I guess it may be because I use the high version python. Could you please tell me what's the version of Python and Perl that seiscomp3 supports? Many thanks.

closed 2018-11-01 by gempa-jabe

samuelchodur, 2020-03-20:
I believe the Python version requirement needs to be reflected in the documentation, unless I am missing something.

Are there plans to move to Python 3 as Python 2 is not longer supported?

jsaul, 2020-03-20:
Sam wrote on 20.03.20 19:24:
> Are there plans to move to Python 3 as Python 2 is not longer supported?

Yes, absolutely. The upcoming major release will have full Python 3 support. Python 2 support will be phased out and limited to 2.7.