SeisComP3/seiscomp3 #171: Install seiscomp on Arch Linux

morealaz, 2018-04-27:
Hi there,
I create a [seiscomp3 package](https://aur.archlinux.org/packages/seiscomp3/) in [Arch Linux User Repository](https://aur.archlinux.org/) for installing seiscomp. User of Arch can use this package to install seiscomp on Arch. feel free to test it.

gempa-jabe, 2018-04-27:
Great! There are a few commits in master that could help with the Python 2/3 issue.

emmaPatagonia, 2018-06-26:
Hi! I am having this problem while compiling seiscomp3 using yaour:

`/tmp/yaourt-tmp-diego/aur-seiscomp3/src/seiscomp3-release-jakarta-2017.334.05/src/trunk/libs/seiscomp3/io/records/sacrecord.cpp: In member function ‘virtual void Seiscomp::IO::SACRecord::write(std::ostream&)’:
/tmp/yaourt-tmp-diego/aur-seiscomp3/src/seiscomp3-release-jakarta-2017.334.05/src/trunk/libs/seiscomp3/io/records/sacrecord.cpp:304:9: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 8 equals destination size [-Wstringop-truncation]
  strncpy(header.knetwk, _net.c_str(), 8);
  ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/tmp/yaourt-tmp-diego/aur-seiscomp3/src/seiscomp3-release-jakarta-2017.334.05/src/trunk/libs/seiscomp3/io/records/sacrecord.cpp:305:9: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 8 equals destination size [-Wstringop-truncation]
  strncpy(header.kstnm, _sta.c_str(), 8);
  ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/tmp/yaourt-tmp-diego/aur-seiscomp3/src/seiscomp3-release-jakarta-2017.334.05/src/trunk/libs/seiscomp3/io/records/sacrecord.cpp:306:9: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 8 equals destination size [-Wstringop-truncation]
  strncpy(header.khole, _loc.c_str(), 8);
  ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/tmp/yaourt-tmp-diego/aur-seiscomp3/src/seiscomp3-release-jakarta-2017.334.05/src/trunk/libs/seiscomp3/io/records/sacrecord.cpp:307:9: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 8 equals destination size [-Wstringop-truncation]
  strncpy(header.kcmpnm, _cha.c_str(), 8);
  ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[  9%] Building CXX object src/trunk/libs/seiscomp3/CMakeFiles/seiscomp3_core.dir/io/records/binaryrecord.cpp.o
[  9%] Building CXX object src/trunk/libs/seiscomp3/CMakeFiles/seiscomp3_core.dir/io/records/ahrecord.cpp.o
/tmp/yaourt-tmp-diego/aur-seiscomp3/src/seiscomp3-release-jakarta-2017.334.05/src/trunk/libs/seiscomp3/io/records/ahrecord.cpp:14:10: fatal error: rpc/rpc.h: No such file or directory
 #include <rpc/rpc.h>
          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [src/trunk/libs/seiscomp3/CMakeFiles/seiscomp3_core.dir/build.make:817: src/trunk/libs/seiscomp3/CMakeFiles/seiscomp3_core.dir/io/records/ahrecord.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3970: src/trunk/libs/seiscomp3/CMakeFiles/seiscomp3_core.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
==> ERROR: Makepkg was unable to build seiscomp3.
==> Restart building seiscomp3 ? [y/N]
==> ----------------------------------
==> 
`

Do you know what is happening ?

gempa-jabe, 2018-06-27:
Archlinux has moved the rpc header to another directory. We have removed the ahrecord.*  files and it should compile again with the master branch. Note that the release 3017.334 will not work. You can try to apply commit 36d0db00709f3cc9c6786499e4ecd56dfa43ea96 to the release source.

emmaPatagonia, 2018-06-27:
Hi gempa, thanks for your kind reply. I note that the only release available in yaour is the 2017.334... I am looking forward to learn how to install seiscomp in arch linux =/

gempa-jabe, 2018-06-28:
The Arch Linux package has been created by @morealaz. We do not support Arch Linux officially because it is a rolling release distribution. You can download packages for CentOS, Ubuntu and other Linux flavours.

morealaz, 2018-06-29:
Arch Linux use [libtirpc](http://libtirpc.sourceforge.net/) to provide necessary libraries and header files for `rpc`. I add it as dependency to Arch [seiscomp3](https://aur.archlinux.org/packages/seiscomp3/) package and fixed its compilation errors. It now compile successfully. @emmaPatagonia please test it, and report any problem here or on AUR. 

I also create a [seiscomp3-git](https://aur.archlinux.org/packages/seiscomp3-git/) package in AUR to install seiscomp3 from latest source and commits from github. You can also try it, if you want to have latest build from github source.

@gempa-jabe: when I tried to compile siescomp from github, I have some errors. I already managed to fixed them, and compile and install seiscomp successfully on Arch. 
First I get following error:

```
[ 35%] Linking CXX shared library ../../../../lib/libseiscomp3_client.so
/usr/bin/ld: /home/morealaz/Projects/Arch/AUR/seiscomp3-git/src/seiscomp3/src/trunk/libs/3rd-party/spread/lib/libtspread-core.a(sp.to): relocation R_X86_64_PC32 against symbol `SP_connect_timeout' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
make[2]: *** [src/trunk/libs/seiscomp3/CMakeFiles/seiscomp3_client.dir/build.make:1138: lib/libseiscomp3_client.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:4132: src/trunk/libs/seiscomp3/CMakeFiles/seiscomp3_client.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```

error suggest that I add `-fPIC` to compiler flags, so I did it. then after that I get following error:

```
[ 35%] Linking CXX executable ../../../../../bin/scmaster
/usr/bin/ld: CMakeFiles/scmaster.dir/master.cpp.o: undefined reference to symbol 'pthread_condattr_setclock@@GLIBC_2.3.3'
/usr/lib/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [src/trunk/apps/messaging/scmaster/CMakeFiles/scmaster.dir/build.make:136: bin/scmaster] Error 1
make[1]: *** [CMakeFiles/Makefile2:1160: src/trunk/apps/messaging/scmaster/CMakeFiles/scmaster.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```

it related to libpthread, and by adding `-lpthread` to linker flags, this error also fixed and seiscomp compile successfully. this error also occured in compiling seiscomp 2017.334.05 and was fixed by adding  `-lpthread` to linker flags.