SeisComP3/seiscomp3 #178: Cache issue in scmag

Brtle, 2018-06-08:
Some times ago, the `removeObject` method have been [implemented in `scamp`](https://github.com/SeisComP3/seiscomp3/issues/152) to allow the removal of amplitude in its cache. Could you implement it for the `scmag` module as well?

I have the same issue that the `scamp` module. I have a custom Python script which remove Magnitude, StationMagnitude and StationMagnitudeContribution but I think the cache keep them in memory. It would be great if the magnitudes could be removed!

gempa-jabe, 2018-06-08:
Refers to #152. Yes, that should be done.

gempa-jabe, 2018-06-08:
Hold on a second ... where do you send the messages to? scmag does not listen for MAGNITUDE group and would not get the messages anyway. What do you expect from scmag once it receives the information about removed magnitudes? To recompute them?

Brtle, 2018-06-08:
I indeed want to recompute the magnitudes. My script looks like:

```python
ep = DataModel.EventParameters()

DataModel.Notifier.Enable()

if self.remove_magnitudes is not None:
    self._remove_magnitudes(origin)

if self.remove_amplitudes is not None:
    self._remove_amplitudes(ep, origin)

DataModel.Notifier.Create(ep, DataModel.OP_UPDATE, origin)

DataModel.Notifier.Disable()

message = DataModel.Notifier.GetMessage(True)
self.connection().send(message)
```

The idea is to remove amplitudes and magnitudes and let `scamp` and `scmag` recompute the new objects. The message is sent to the LOCATION group.

However, if `scamp` isn't enable or doesn't calculate a new amplitude, the old cached magnitude will still exist in scmag.