SeisComP3/seiscomp3 #189: seiscomp3 How to add external dynamic library dependencies?

WuFhong, 2018-08-15:
Download the build version, the project does not have a pro file, how to add an external dependency library? Add to CmakeList? The path to the spread call library was not found.Ask the Great God to tell,thanks!

gempa-jabe, 2018-08-15:
Sorry, but be more specific on what you try to do. The build is based on cmake and not on pro files. Spread is included in the sources and not an external library.

WuFhong, 2018-08-16:
I want to change spread, my own middleware dynamic library file not to add, I will not add in Cmake, and spread does not have main function, can't add project name when LINK, can you teach me? Thank you very much.

WuFhong, 2018-08-16:
I want to replace the spread, my own middleware dynamic library files not added, I will not add to the Cmake, and spread has no main function, can not add the name of the project when LINK, can you teach me? Thank you very much.

WuFhong, 2018-08-16:
I want to change spread, my own middleware dynamic library file add not successful, I will not add in Cmake, and spread does not have main function, can't add project name when LINK, can you teach me? Thank you very much.

WuFhong, 2018-08-16:
Spread is included in the sources and not an external library. I'm sorry, I don't understand this? Can you explain in detail?

WuFhong, 2018-08-16:
Where is the spread dynamic library. So file linked? Please tell me. Thank you very much.

WuFhong, 2018-08-16:
Where do I link the dynamic library of spread? It's libspread.so. please tell me, thank you very much.

gempa-jabe, 2018-08-16:
Do you want to develop your own software based on this code or do you just want to compile it?

If you want to develop your own code, you have to link your application against libseiscomp3_client. This also links against Spread. Again, what do you want to do and how and what does not work? Give an example or some code otherwise I cannot help you.

WuFhong, 2018-08-16:
I want to replace spread with my own middleware, now my own middleware dynamic library can not link to the project
error is : No rule to make targat "src/trunk/lib/core_reges.so", needed by "src/link/libs/seiscomp3/CMakeFiles/seiscomp3_client.dir/communication/spread/spreaddriver.cpp.o".Stop

gempa-jabe, 2018-08-16:
How does your CMakeLists file look like and where is your library located?

WuFhong, 2018-08-16:
I added the link .So library in cmakelist.txt under src/trunk/libs/seiscomp3 in the libs folder under home/console

[image]

gempa-jabe, 2018-08-16:
Remove all the BITSFLOW definitions because it is not build within the source. If it is an external library, then just add it to SC_LINK_LIBRARIES at the bottom. That's it. You should better add another driver for your bitsflow middleware instead of hijacking the Spread driver.

WuFhong, 2018-08-16:
What if I want to replace spread?What should I do?

gempa-jabe, 2018-08-16:
I would advise to start with a new driver, test it and when it is working and when you understood how the system works, start to replace Spread if really necessary at your own risk. Replacing Spread is in general a bad option because you have to patch our sources again and again. Do it like the httpmsgbus driver, it is a very good example on how to support additional message buses.

Just add a new directory, e.g. ```src/trunk/libs/seiscomp3/communication/bitsflow```.
Add two new files to it that implement the middleware driver: ```src/trunk/libs/seiscomp3/communication/bitsflow/bfdriver.h```, ```src/trunk/libs/seiscomp3/communication/bitsflow/bfdriver.cpp```.

Add to ```src/trunk/libs/seiscomp3/communication/CMakeLists.txt```:

```
COM_SOURCES: ... bitsflow/bfdriver.cpp
COM_HEADERS: ... bitsflow/bfdriver.h
```

Add to ```src/trunk/libs/seiscomp3/CMakeLists.txt```:

```
SC_LIB_LINK_LIBRARIES(client
	...
	bitsflowlib
)
```

Good luck!

WuFhong, 2018-08-16:
Would it be possible if I were to replace the spread completely?Thank you very much indeed.

gempa-jabe, 2018-08-16:
Yes, it is possible but I cannot help you through the entire process as it is not our interest. I recommend to write and test an additional driver and then remove Spread. Removing Spread is straightforward: just remove all references to it and fix the sources that will not compile anymore.

WuFhong, 2018-08-16:
OK, thank you very much for your answer. May seiscomp3 be better and better.

gempa-jabe, 2018-08-16:
One last thing: if you remove Spread then you will not be able to connect to other SC3 instances that still run Spread. The advantage of a new driver is that you can support both. In the configuration you can either use:

```
# Connects to Spread. The default protocol is spread and the
# line below is equal to "spread://localhost"
connection.server = localhost
```

or

```
connection.server = bitsflow://localhost # Connects to bitsflow server
```

That could be important for e.g. scimport to import message from a Spread system into your bitsflow system: spread://remove-host -> bitsflow://localhost

WuFhong, 2018-08-16:
Is it convenient for you to tell me how to test the httpdriver driver?

gempa-jabe, 2018-08-16:
See https://forum.seiscomp3.org/t/using-scimex-with-httpmsgbus/85. For further questions regarding httpmsgbus please use the forum and comment there.

closed 2018-08-16 by WuFhong