SeisComP3/seiscomp3 #190: seiscomp3 FDSNxml to  sc3xml conversion: m/s Vs M/S units  - potential Issue

salichon, 2018-08-16:
Hi
Using the inventory scheme 0.1 and seiscomp3 version 2017.334.p5.
and converting a FDSN station xml  into a seiscomp3 inventory xml 

Having the unit m/s in the seiscomp3 xml in lower capital prevents a "secondary" stream to be displayed in the GUI scolv
 [EAZ_fdsnZ_work.xml.txt](https://github.com/SeisComP3/seiscomp3/files/2292266/EAZ_fdsnZ_work.xml.txt)

As per example (only one component) below that is displaying the unpicked  secondary stream (BNZ) and the HHZ picked stream: 
Here attached the expected display as an example.  
 [image]

and the one using the m/s
[image]

Cheers
jerome

gempa-jabe, 2018-08-16:
The lowercase issue is going to be fixed. Thanks for the report Jerome.

gempa-jabe, 2018-08-16:
I hope that I fixed it. Can you test 9beb44888c36d6e14f3d5cdbb6cf89427fbbeed6?

salichon, 2018-08-17:
Hello Jan  i will next week
Cheers!
JErome

quiffman, 2018-08-24:
Hi @gempa-jabe!

@salichon didn't have time to check this before he left - so I just did a quick recompile to test and it looks right to me...

[image]