SeisComP3/seiscomp3 #194: [fdsnws]: Perfomance issue for fdsnws/station

damb, 2018-10-02:
Hi devs,

I'm one of the guys developing the new [`eida-federator`](https://github.com/EIDA/mediatorws) aiming at providing distributed and federated (meta)data access within [EIDA](https://www.orfeus-eu.org/data/eida/). Almost every datacenter (except of [RESIF](https://www.resif.fr/?lang=en)) uses webservices from SC3. `eida-federator` implements the Interfaces of `fdsnws-station`, `fdsnws-dataselect` and `eidaws-wfcatalog`.

Of course, there are different approaches when requesting data from endpoints. In general `eida-federator` implements an approach requesting data from endpoints based on stream level granularity (in order to provide granular, stream epoch based logging facilities in future). For station metadata, however, this approach seems not feasible. Requesting station metadata based on stream level granularity simply takes too long (several minutes). For example imagine a request like:
```
http://federator-testing.ethz.ch/fdsnws/station/1/query?level=channel&format=text
```
Once received a request as above  `eida-federator` will issue requests for every single stream epoch to every single endpoint (i.e. `fdsnws-station`) within EIDA and finally return a merged result to the requesting client.

That is why during our last ETC meeting at Grenoble for station metadata federation of not distributed physical networks we decided to switch to an approach using *bulk requests* based on network code granularity. This means, that e.g. for a request like
```
http://federator-testing.ethz.ch/fdsnws/station/1/query?net=ZS&level=channel&minlat=45&maxlat=47&minlon=10&maxlon=14
```
with our *new* approach `eida-federator` issues the following request to the `fdsnws-station` endpoint GFZ (see attachment). Note, that all stream epochs are explicit (i.e. no wildcards). 
Unfortunately some `fdsnws-station` SC3 webservice installations (e.g. GFZ) are not able ton respond this request within 60s such that either HTTP status code 504 or 502 is returned.

There is currently an issue (EIDA/mediatorws#32) discussing approaches how to proceed (for federated `fdsnws-station` requests). Besides of optimizing the implementation of `eida-federator` of course there is also space to improve the endpoints. To sum the suggested `fdsnws-station` SC3 endpoint related topics up:
- As @andres-h pointed out a copy of the inventory data is kept in RAM such that the DB isn't accessed at all while handling requests. However, with this approach the inventory data is not structured such that your lookup is really inefficient. Besides the lookup is implemented in Python.  As a result, the current SC3 `fdsnws-station` implementation is simply too slow.
- Do you see any possibility accessing the DB directly / using an ORM framework? (I didn't check so far if something like a (handcrafted) ORM framework is already shipping with the SC3 C++ core. In this case we could make use of `SWIG`.)
- When using a *bulk request* approach based on network code granularity the [generator](https://github.com/SeisComP3/seiscomp3/blob/9b22e3309a5e2559623b8f80c182397f1b62d4c9/src/trunk/apps/fdsnws/fdsnws/station.py#L122-L140) yielding networks emerges only a single network. Hence, from the SC3 `fdsnws-station` endpoint's perspective this type of request should be already quite cheap. However, for `level=channel|response` still lookups for station, location and stream codes are performed.

In future we also plan to perform load/performance tests (both for `eida-federator` and for endpoints). However, as long as the limitation is a single `fdsnws-station` request IMO we should first discuss and focus on the points mentioned above.

cheers

Attachments:
* [ZS.txt](https://github.com/SeisComP3/seiscomp3/files/2434519/ZS.txt)

gempa-jabe, 2018-10-02:
Interesting. Please note that we are not involved in EIDA in any way so I cannot and will not touch any of the points related to that. So it boils down to "fdsnws-station is too slow for the attached example request at GFZ".

We can test that request with a local installation. It is correct that the inventory is held in memory to avoid expensive database accesses. And even if the local lookups are implemented in Python, I cannot imagine that the request causes the response to take 1 minutes or more. That sounds too much, I agree. So we have to identify the bottleneck first before thinking about solutions.

> When using a bulk request approach based on network code granularity the generator yielding networks emerges only a single network. Hence, from the SC3 fdsnws-station endpoint's perspective this [...]

Note that possibly access rules are defined outside the database. So fdsnws has to check any object with respect to the request IP or user.

We will test your request and come back with numbers.

jsaul, 2018-10-02:
Note that the fdsnws-station at GFZ was down from about 12:30 p.m. UTC yesterday until about 2:00 a.m. UTC today, returning only 502 errors. Something was seriously screwed up, either due to excessive testing activities or 3rd party requests. This needs to be investigated in depth. Please do not publicly make claims about "SC3 webservice installations not able to respond within 60s". Thank you.

damb, 2018-10-02:
@gempa-jabe, I did not expect your answer. But, I should have tested once again before posting this issue. The GFZ behaviour from yesterday afternoon and the 2018/09/26 is not reproducible anymore.

I must confirm that the exemplary request from above is returned in ~ **2.5s** using `curl`.

@jsaul:

> Note that the fdsnws-station at GFZ was down from about 12:30 p.m. UTC yesterday until about 2:00 a.m. UTC today, returning only 502 errors. Something was seriously screwed up, either due to excessive testing activities or 3rd party requests. This needs to be investigated in depth.

Thanks for this information. Perhaps the `eida_maint@gfz-potsdam.de` mailing list would have been a better place to discuss this issue.

Sorry for the inconvenience.

You can close this issue.

closed 2018-10-02 by gempa-jabe