SeisComP3/seiscomp3 #196: numeric characters in network code

cweidle, 2018-10-05:
When running scautoloc with picks from stations with network code 5H I run into:
`17:26:59 [error] Autoloc3: MISSING STATION 5H.COO05`
In an scxmldump -fI, the stations of this network are contained, thus I assume that the inventory is fine.

I have a playback setup where I write picks to xml files and then call scautoloc
`seiscomp exec scautoloc --ep pick.xml -d $host --station-config mystation.conf --debug > origins.xml` 

mystation.conf is
> \* \* 0 60
> 5H \* 1 60
> II \* 1 180

From the debug output I get the feeling that only stations with [A-Z] network codes are read.
> 17:26:58 [debug] Reading station locations from DB
> 17:26:58 [debug] Initialized station II.RAYN
> 17:26:58 [debug] Initialized station II.UOSS

the same when reading `mystation.conf`
> 17:26:58 [debug] Reading station configs from file 'mystation.conf'
> 17:26:58 [debug] Station II RAYN   pattern II *     config: usage=1 maxnucdist=180
> 17:26:58 [debug] Station II UOSS   pattern II *     config: usage=1 maxnucdist=180

in both cases, stations from network 5H are missing.

Also, when I run scolv, the stations from network II are shown on the map, but network 5H is missing.
Since this is an official network code from FDSN I would like to avoid renaming it.
Cheers
Christian

cweidle, 2018-10-05:
Just see that this might be the same as #104 as the stations are offline and have an END in the inventory.

jsaul, 2018-10-11:
I committed a fix to the issue. Please give it a try.