SeisComP3/seiscomp3 #203: nmxptool installing

barbod2543, 2018-11-21:
Hi dears
When i use “make” for compiling this error appear:

[mehrdad@localhost nmxptool-2.1.4]$ ./configure
checking build system type… x86_64-unknown-linux-gnu
checking host system type… x86_64-unknown-linux-gnu
checking target system type… x86_64-unknown-linux-gnu
configure: BINARY_DIST_NAME_SUFFIX set to x86_64-unknown-linux-gnu
checking for a BSD-compatible install… /usr/bin/install -c
checking whether build environment is sane… yes
checking for a thread-safe mkdir -p… ./install-sh -c -d
checking for gawk… gawk
checking whether make sets $(MAKE)… yes
checking whether make supports nested variables… yes
checking whether to enable maintainer-specific portions of Makefiles… no
checking for gawk… (cached) gawk
checking for gcc… gcc
checking whether the C compiler works… yes
checking for C compiler default output file name… a.out
checking for suffix of executables…
checking whether we are cross compiling… no
checking for suffix of object files… o
checking whether we are using the GNU C compiler… yes
checking whether gcc accepts -g… yes
checking for gcc option to accept ISO C89… none needed
checking whether gcc understands -c and -o together… yes
checking for style of include used by make… GNU
checking dependency style of gcc… gcc3
checking whether ln -s works… yes
checking whether make sets $(MAKE)… (cached) yes
checking how to print strings… printf
checking for a sed that does not truncate output… /usr/bin/sed
checking for grep that handles long lines and -e… /usr/bin/grep
checking for egrep… /usr/bin/grep -E
checking for fgrep… /usr/bin/grep -F
checking for ld used by gcc… /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld… yes
checking for BSD- or MS-compatible name lister (nm)… /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface… BSD nm
checking the maximum length of command line arguments… 1572864
checking how to convert x86_64-unknown-linux-gnu file names to x86_64-unknown-linux-gnu format… func_convert_file_noop
checking how to convert x86_64-unknown-linux-gnu file names to toolchain format… func_convert_file_noop
checking for /usr/bin/ld option to reload object files… -r
checking for objdump… objdump
checking how to recognize dependent libraries… pass_all
checking for dlltool… no
checking how to associate runtime and link libraries… printf %s\n
checking for ar… ar
checking for archiver @FILE support… @
checking for strip… strip
checking for ranlib… ranlib
checking command to parse /usr/bin/nm -B output from gcc object… ok
checking for sysroot… no
checking for a working dd… /usr/bin/dd
checking how to truncate binary pipes… /usr/bin/dd bs=4096 count=1
checking for mt… no
checking if : is a manifest tool… no
checking how to run the C preprocessor… gcc -E
checking for ANSI C header files… yes
checking for sys/types.h… yes
checking for sys/stat.h… yes
checking for stdlib.h… yes
checking for string.h… yes
checking for memory.h… yes
checking for strings.h… yes
checking for inttypes.h… yes
checking for stdint.h… yes
checking for unistd.h… yes
checking for dlfcn.h… yes
checking for objdir… .libs
[... 157 lines cut ...]
mv -f .deps/nmxptool-nmxptool_listen.Tpo .deps/nmxptool-nmxptool_listen.Po
gcc -DHAVE_CONFIG_H -I. -I… -I…/include -g -O2 -MT nmxptool-seedlink_plugin.o -MD -MP -MF .deps/nmxptool-seedlink_plugin.Tpo -c -o nmxptool-seedlink_plugin.o test -f 'seedlink_plugin.c' || echo './' seedlink_plugin.c
mv -f .deps/nmxptool-seedlink_plugin.Tpo .deps/nmxptool-seedlink_plugin.Po
/bin/sh …/libtool --tag=CC --mode=link gcc -I…/include -g -O2 -o nmxptool nmxptool-nmxptool.o nmxptool-nmxptool_getoptlong.o nmxptool-nmxptool_ew.o nmxptool-nmxptool_chanseq.o nmxptool-nmxptool_sigcondition.o nmxptool-nmxptool_listen.o nmxptool-seedlink_plugin.o …/lib/libnmxp.a -lrt -lm
…/libtool: line 7902: func_munge_path_list: command not found
libtool: link: gcc -I…/include -g -O2 -o nmxptool nmxptool-nmxptool.o nmxptool-nmxptool_getoptlong.o nmxptool-nmxptool_ew.o nmxptool-nmxptool_chanseq.o nmxptool-nmxptool_sigcondition.o nmxptool-nmxptool_listen.o nmxptool-seedlink_plugin.o …/lib/libnmxp.a -lrt -lm
/usr/bin/ld: nmxptool-nmxptool.o: undefined reference to symbol ‘pthread_create@@GLIBC_2.2.5’
//usr/lib64/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [nmxptool] Error 1
make[2]: Leaving directory /home/mehrdad/seiscomp3/share/plugins/seedlink/nmxptool-2.1.4/src' make[1]: *** [all-recursive] Error 1 make[1]: Leaving directory /home/mehrdad/seiscomp3/share/plugins/seedlink/nmxptool-2.1.4’
make: *** [all] Error 2
[mehrdad@localhost nmxptool-2.1.4]$

Reply

barbod2543
40m
plz answer me for this error:pensive::pensive::pensive::pensive:
/usr/bin/ld: nmxptool-nmxptool.o: undefined reference to symbol ‘pthread_create@@GLIBC_2.2.5’
//usr/lib64/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [nmxptool] Error 1
make[2]: Leaving directory /home/mehrdad/seiscomp3/share/plugins/seedlink/nmxptool-2.1.4/src' make[1]: *** [all-recursive] Error 1 make[1]: Leaving directory /home/mehrdad/seiscomp3/share/plugins/seedlink/nmxptool-2.1.4’
make: *** [all] Error 2

gempa-jabe, 2018-11-21:
nmxptool is not part of SeisComP3, it is an external application. Please contact the author of the tool. Maybe "-lpthread" is missing in the linker command line?

closed 2018-11-21 by gempa-jabe

barbod2543, 2018-11-21:
in which file?

jsaul, 2018-11-21:
Which part of "Please contact the author of the tool." did you not understand?

gempa-jabe, 2018-11-21:
Eventually in the Makefile. The project is using autotools and SeisComP is using cmake, completely different build systems. Again, please contact the author to resolve the compile issue.

gempa-jabe, 2018-11-21:
You can create an issue at the nmxptool project page: https://gitlab.rm.ingv.it/matteo.quintiliani/nmxptool