SeisComP3/seiscomp3 #207: Out-of-order packets handling in SCOLV

rannof, 2018-12-04:
We've encountered different behaviour of SCOLV's picker GUI when using archived data containing out-of-order packets. When using sdsarchive as input source, out-of-order packets (or overlaps) are discarded and marked as gaps, while when using the file itself (i.e. -I file://... command-line option) traces include no gaps on the SCOLV picker GUI.
1. I would like to verify this behaviour.
2. I would like to check if there is any mechanism (beside the obvious re-ordering the data) within SCOLV or sdsarchive to set out-of-order packets handling such that no gaps will appear. Since this mechanism is (apparently) available for the -I option, would it be possible to optionally enable it for the sdsarchive too? I believe there might be an undesired overhead for the reordering but if its an optional feature, it might solve some issues for others too.

Cheers,
Ran

gempa-jabe, 2018-12-04:
scolv always re-orders packages. The issue here is that the sdsarchive handler skips out-of-order packages. scolv will never receive them. The file handler forwards everything without further checking, that's why you observe the correct behaviour with ```file://```.

An option to solve that is to add an option to the sdsarchive handler, e.g. ```sdsarchive:///path/to/archive?ooo=true```. And yes, it can be more expensive as one would have to search the entire file for any requested time window. In a perfect world one would even re-order the packages before delivering them to the client and make that the default behaviour. I would prefer the latter option but can't promise anything. As always, PRs are welcome.

rannof, 2018-12-04:
Thank you for the (as always) quick reply.
Does the ooo=true option already implemented? or is it a suggestion for a future feature?

gempa-jabe, 2018-12-04:
Both are suggestions for a future feature. I would like to focus on the second. Actually a user does not want to take care of such technical details. I would prefer to allow disabling re-ordering as optimization for power users who know exactly how their archive is structured.

rannof, 2018-12-04:
Looking forward for this feature. Any suggestions for a quick fix for now?

gempa-jabe, 2018-12-04:
Other than fixing code? No.

rannof, 2018-12-18:
Hi,
Our IT expert Andrey Polozov (andre.user12@gmail.com) have created a working plugin for reading the archived files into memory (one by one), reorder and slice the requested span.
Unfortunately, we are not familiar enough with the PR system, but the attached are the plugin code and header to be compiled similar to the original sdsarchive plugin. The files are lacking comments, but we hope it might serve as a possible solution to anyone out there until a more formal solution would be introduced.
Please let us know if there are any problems with the code.

[sdsarchsort.zip](https://github.com/SeisComP3/seiscomp3/files/2690766/sdsarchsort.zip)

gempa-jabe, 2018-12-20:
Thank you very much for the code, well done.

A reason why we can't use it as is is that it is using C++ 11 which we still do not use (will come). We will also need to test it. The approach is reasonable to read the entire file in memory, sort it and read from it. There can be cases when memory comes into play. The sdsarchive source is also used in fdsnws and with many concurrent request it could use lots of memory. The good thing is that you added another source "sdsarchsort", so it does not interfere with the current implementation. Eventually I would like to see an almost perfect sdsarchive implementation which handles out-of-order files in a performant and memory friendly way so that users do not need to decide which implementation is the right one for them.

For the time being I recommend that you are using that code as plugin or as a patch in your environment. Is that feasible?

Thanks again,
Jan

closed 2020-12-07 by gempa-jabe