SeisComP3/seiscomp3 #213: Question: ZRT waveform data projection

luca-s, 2019-01-03:
Dear developers,

I would like to ask if seiscomp3 code base provides already API for waveform data projection into ZRT domain, the same way as the Picker GUI does (there it is possible to easily switch between Z12, ZNE, ZRT etc.). I need to do the same and I would like to avoid writing the projection code if that is already available in seiscomp3.

Thank you!

luca-s, 2019-01-15:
Is forum.seiscomp3.org a better place for this kind of question? I was not sure so I posted the question here

gempa-jabe, 2019-01-15:
Actually there is a software development channel in the forum. This is an issue tracker. I can try to reply here anyway:

It depends on the component where you would like to rotate the data. Do you want to do it in Python or C++? The GUI provides a specialized 3C trace which supports arbitrary rotations, the core supports the same but it is a bit more difficult to use it. And it only works from C++.

luca-s, 2019-01-15:
I am using C++. Does the GUI have an improved API built on top of the core API or is it completely independent? May I ask where the core API for projection reside? 

thank you.

luca-s, 2019-01-15:
Ah, I have now seen the development channel in the forum, thank you. I will use that next time

gempa-jabe, 2019-01-15:
There are some additional classes especially used with some widgets that have similar code in the GUI to that that exists in the core. Actually you are right that the GUI libs are built on top of the core.

The code resides in ```src/trunk/libs/seiscomp3/processing/operator/ncomps.h```.

The NCompsOperator takes an arbitrary number of channels, synchronizes them and runs a processor on common time windows as soon as they are available. To buffer individual channels you can configure the buffer size, the default buffers without limit. Code example:

```c++
using namespace Seiscomp;
using namespace Seiscomp::IO;
using namespace Seiscomp::Processing;

bool store(const Record *rec) {
	MSeedRecord mseed(*rec);
	mseed.setLocationCode("11");
	mseed.write(cout);
	cerr << "  >  " << rec->streamID() << " "
	     << rec->startTime().toString("%T.%f") << " "
	     << rec->endTime().toString("%T.%f") << endl;
	return true;
}

Stream streams[3];
// Based on the channel code only the input data is directed into the right slots.
streams[0].setCode("HHZ");
streams[0].gain = 2.0;
streams[1].setCode("HHN");
streams[1].gain = 3.0;
streams[2].setCode("HHE");
streams[2].gain = 4.0;

// The wrapper will direct 3 codes into the right slots using the
// Stream configuration class and will finally use the transformation
// operator. The advantage is that it will apply the configured gain for. If you don't need
// that you can replace it with CodeWrapper.
typedef Operator::StreamConfigWrapper<float,3,Operator::Transformation> OpWrapper;

// Define the final operator class:
//  1. Send channel codes to right slots
//  2. Align 3 channels sample wise
//  3. Transform the resulting 3 component trace with a rotation matrix
typedef NCompsOperator<float,3,OpWrapper> Rotator;

Math::Matrix3f m;
// TODO: Populate matrix m

Rotator op(OpWrapper(streams, Operator::Transformation<float,3>(m)));

// The function that will be called after a transformed record was created
op.setStoreFunc(store);

RecordInput inp(stream.get(), Array::FLOAT);
for ( RecordIterator it = inp.begin(); it != inp.end(); ++it ) {
	op.feed(*it);
}
```

Usually the WaveformOperator is used in combination with WaveformProcessor but can also be used without it as you can see.

**Note** that the Transformation processor currently has an incomplete interface and will not compile. I have pushed a fix and you can easily copy the new implementation if you need to build against an older version of SC3.

I hope that helps. Let me know if you have further questions.

luca-s, 2019-01-15:
Thank you very much @gempa-jabe . I will experiment with the code and if I have further questions I will open a thread on the forum.

closed 2019-01-15 by luca-s