SeisComP3/seiscomp3 #215: Seiscomp3 compilation on MacOsX problem linking cxx library

Radiasan, 2019-01-11:
Dear users, 

I have a problem when compiling Seiscomp3 on MacOS 10.13.6. This is what i am getting: 
------------------------------------------
  File "<string>", line 6
    print 'Byte-compiling python modules...'
                                           ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print('Byte-compiling python modules...')?
[ 16%] Building C object src/seedlink/plugins/fs_plugin/CMakeFiles/fs_plugin.dir/__/plugin.c.o
make[2]: *** [lib/_Config.so] Error 1
make[2]: *** Deleting file `lib/_Config.so'
make[1]: *** [src/system/libs/swig/CMakeFiles/_Config.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 16%] Linking CXX executable ../../../../bin/fs_plugin
[ 16%] Built target fs_plugin
make: *** [all] Error 2

gempa-jabe, 2019-01-15:
You need Python2, not Python3.

gilcel, 2019-01-21:
You can compile the latest version of Seiscomp3 Jakarta for macOS from here:
https://github.com/gilcel/seiscomp3-macos

Some instructions here:
https://github.com/gilcel/seiscomp3-macos/blob/seiscomp3-jakarta-macos-latest/README-macOS.md

Radiasan, 2019-01-21:
> You need Python2, not Python3.

Thanks a lot for you answer. I already have python 2.

Radiasan, 2019-01-21:
> You can compile the latest version of Seiscomp3 Jakarta for macOS from here:
> https://github.com/gilcel/seiscomp3-macos
> 
> Some instructions here:
> https://github.com/gilcel/seiscomp3-macos/blob/seiscomp3-jakarta-macos-latest/README-macOS.md

Thanks a lot for your answer i will try this version.

Radiasan, 2019-01-21:
> You need Python2, not Python3.

Thanks a lot for your reply i already have Python2 installed. I will try the version proposed by Gilcel.

gilcel, 2019-01-22:
I just updated the SEISCOMP3-macOS port version. I forgot to copy the spread lib (and headers) to $SEISCOMPDIR directory.

This is now fixed, here:
https://github.com/gilcel/seiscomp3-macos/commit/7adafb0706116933e5765b949b9e41937e097bea

closed 2019-09-28 by gempa-jabe