SeisComP3/seiscomp3 #218: 32-bit _oid fields overflow

Jollyfant, 2019-03-12:
Hi, our SC3 database ran out of possible `Object._oid` values since this field is limited to a signed 32 bit number. Our last entry is `2147483641` which is just short of `2 ** 31`.

    2019/03/11 15:11:48 [warning/DbPlugin] Error handling message from 7781arc to LOGGING
    2019/03/11 15:11:48 [error/MYSQL] execute("insert into Object(_oid) values(default)") = 167 (Out of range value for column '_oid' at row 1)
    2019/03/11 15:11:48 [error/DatabaseArchive] writing object failed

Does anyone have any suggestions on how to save the database since nothing can be added. We considered updating the field to allow for 64-bit values but it seems the code is not designed to deal with this. Hope you have some advice for us!

In any case this field should be an unsigned integer and perhaps allow larger values?

gempa-jabe, 2019-03-12:
What do you mean with "save"? Backup?

To increase the oid to 64 bit would also need to increase the C++ counterparts in DatabaseArchive. That is currently not the issue.

Jollyfant, 2019-03-12:
With "save" I mean commit a change so that SeisComP3 can save objects again (e.g. changing the field type to `BIGINT`).

> That is currently not the issue.

Sorry I do not understand.

Thx!

gempa-jabe, 2019-03-12:
> Sorry I do not understand.

I meant "not the case" ;)

gempa-jabe, 2019-03-12:
Changing to BIGINT or some other > 32bit type is not supported by the code as you have noticed already, that is what I wanted to say. Sorry for confusion.

andres-h, 2019-03-12:
I created a pull request at https://github.com/SeisComP3/seiscomp3/pull/219

@Jollyfant please check if this works with your huge database.

gempa-jabe, 2019-09-28:
This issue will be addressed in an upcoming release.

closed 2019-09-28 by gempa-jabe