SeisComP3/seiscomp3 #230: [fdsnxml2inv] Conversion to FDSNXML duplicates Datalogger gain

jordi-domingo, 2019-06-26:
If the Datalogger gain is different than 0 (i.e.: pre-amplifier), the conversion from SC3's inventory to FDSNXML duplicates that gain when adding the extra VOLTS to COUNTS stage.

It is likely due to this part:
https://github.com/SeisComP3/seiscomp3/blob/master/src/trunk/apps/tools/inventory/fdsnxml2inv/convert2fdsnxml.cpp#L1030

For instance, if I have an input FDSNXML file with this response:
```
Channel Response
	From M/S (Velocity in Meters per Second) to COUNTS (Digital Counts)
	Overall Sensitivity: 1.9024e+10 defined at 1.000 Hz
	7 stages:
		Stage 1: PolesZerosResponseStage from M/S to V, gain: 1189
		Stage 2: PolesZerosResponseStage from V to V, gain: 40
		Stage 3: CoefficientsTypeResponseStage from V to COUNTS, gain: 400000
		Stage 4: CoefficientsTypeResponseStage from COUNTS to COUNTS, gain: 1
		Stage 5: CoefficientsTypeResponseStage from COUNTS to COUNTS, gain: 1
		Stage 6: CoefficientsTypeResponseStage from COUNTS to COUNTS, gain: 1
		Stage 7: CoefficientsTypeResponseStage from COUNTS to COUNTS, gain: 1
```
And I add it to SC3, and then get the inventory back in FDSNXML, I got this other response:
```
Channel Response
	From M/S () to COUNTS ()
	Overall Sensitivity: 1.9024e+10 defined at 1.000 Hz
	8 stages:
		Stage 1: PolesZerosResponseStage from M/S to V, gain: 1189
		Stage 2: PolesZerosResponseStage from V to V, gain: 40
		Stage 3: CoefficientsTypeResponseStage from V to COUNTS, gain: 40
		Stage 4: FIRResponseStage from COUNTS to COUNTS, gain: 400000
		Stage 5: FIRResponseStage from COUNTS to COUNTS, gain: 1
		Stage 6: FIRResponseStage from COUNTS to COUNTS, gain: 1
		Stage 7: FIRResponseStage from COUNTS to COUNTS, gain: 1
		Stage 8: FIRResponseStage from COUNTS to COUNTS, gain: 1
```
The most important issue is that the gain of 40 is duplicated.

Additionally, I don't understand why an extra stage is added. When the gain is 1, there are no relevant consequences, but it would be nice that the conversion FDSNXML -> SC3ML -> FDSNXML was transparent...

jordi-domingo, 2019-06-26:
Maybe related to this other issue in the past...?

gempa-jabe, 2019-07-12:
Can you please upload an example to reproduce and fix it?

jordi-domingo, 2019-07-12:
I can't provide it right now, but if you create an inventory file with the following datalogger, you can reproduce it:

Nanometrics Centaur High Gain 1Vpp (gain 40)
http://ds.iris.edu/NRL/dataloggers/nanometrics/centaur/RESP.XX.NN391..BHZ.CENTAUR.40.40.OFF.LP

gempa-jabe, 2019-07-12:
I don't have tools at hand to do that. If possible provide me with an example XML file whenever it is convenient for you.

jordi-domingo, 2019-07-12:
This should work (I hope...) 
[NL.TERZ.00.BH.zip](https://github.com/SeisComP3/seiscomp3/files/3386696/NL.TERZ.00.BH.zip)

gempa-jabe, 2019-07-12:
Yes, that shows the issue but it is already SC3 XML. Can you provide the input StationXML file?

I have tried to create a similar StationXML file (see attached) and both conversions look OK. Which SC3 version have you used?

[dev-fdsnws.zip](https://github.com/SeisComP3/seiscomp3/files/3386946/dev-fdsnws.zip)

jordi-domingo, 2019-07-12:
I'll provide the input Station XML

jordi-domingo, 2019-07-12:
I'll provide the StationXML when I'm able to.

However, I think the problem is the output StationXML. If you export that inventory via fdsnws (StationXML), then the extra stage+gain is added.

jordi-domingo, 2019-07-12:
(I closed the issue by mistake...)

jordi-domingo, 2019-07-12:
I've tried it with both version 2017.124 and 2018.327.

gempa-jabe, 2019-07-15:
> However, I think the problem is the output StationXML. If you export that inventory via fdsnws (StationXML), then the extra stage+gain is added.

That is only a problem because the SC3 data is already wrong. The conversion from StationXML to SC3 XML introduced that extra stage. You can try the XML that I added above with the latest SC3 version and some older version, e.g. the one you are running on the server which converted your StationXML. The output should be different. SC3 XML to StationXML conversion cannot be fixed. The input (StationXML -> SC3) must updated and then it should work.

jordi-domingo, 2019-07-15:
Ok, thanks for the clarification! I'll try it when I'm back at the office.

jordi-domingo, 2019-08-28:
Hi @gempa-jabe

jordi-domingo, 2019-08-28:
Hi @gempa-jabe 
You were right, the problem was on the conversion StationXML -> SC3.
With a newer version of SC3 (2018.327) it works well, and the output of FDSNws in StationXML format doesn't have the doubled gain.
Thanks!

closed 2019-08-28 by jordi-domingo