SeisComP3/seiscomp3 #237: FDSNWS: remove unsupported matchtimeseries parameter from station wadl

megies, 2019-07-22:
It seems that the `matchtimeseries` parameter is unsupported by the FDSN station WS but it is in the application wadl and should be removed: https://github.com/SeisComP3/seiscomp3/blob/911587c100b650caf2084a6de93ccec232e72bb0/src/trunk/apps/fdsnws/share/station.wadl#L33

See EIDA/userfeedback#24

gempa-jabe, 2019-07-22:
I have tried the links in the referenced issue. If matchtimeseries is set to false then it returns the list of channels. If it is set to true it returns 204 (no content). It is evaluated correctly. Both responses are perfectly valid. Why do you think that it is not supported? Do I miss something important?

megies, 2019-07-22:
Well, the thing is, if `matchtimeseries=true` is used we seem to always get an empty station list, even if there is data:

- http://erde.geophysik.uni-muenchen.de/fdsnws/station/1/query?station=FFB*&starttime=2019-07-01T00%3A00%3A00.000000&endtime=2019-07-02T00%3A00%3A00.000000&level=channel&format=text
- http://erde.geophysik.uni-muenchen.de/fdsnws/station/1/query?station=FFB*&starttime=2019-07-01T00%3A00%3A00.000000&endtime=2019-07-02T00%3A00%3A00.000000&level=channel&format=text&matchtimeseries=true
- http://erde.geophysik.uni-muenchen.de/fdsnws/dataselect/1/query?station=FFB*&starttime=2019-07-01T12%3A00%3A00.000000&endtime=2019-07-01T12%3A00%3A05.000000

gempa-jabe, 2019-07-22:
OK. Getting empty data is a different thing than "the parameter is not supported". To make matchtimeseries work as expected you need to populate the data availability database. That is a feature that has been introduced recently. Without the database availability database the service cannot match time series with meta data and therefore returns empty content. In SC3 we have a tool called scardac that creates this database from an SDS archive.

Without the data availability feature (`dataAvailability.enable = false`, the default is `true`) the existence of the matchtimeseries parameters would return an error. We would prefer to change it the way that it would just ignore the parameter if the data availability feature is disabled. Would you agree?

megies, 2019-07-22:
I agree that ignoring the parameter makes sense if the feature is disabled internally. Ideally the wadl should be constructed dynamically as well I guess, depending on whether the feature is enabled or not.

gempa-stephan, 2019-07-23:
In #238 I implemented a dynamically generated WADL file. With `dataAvailability` disabled the `matchtimeseries` parameter is omitted from the WADL file. Also the default value for `dataAvailability` was changed from `true` to `false`.

megies, 2019-07-24:
Good to hear @gempa-stephan, I think it makes sense this way. :+1:

closed 2019-09-28 by gempa-jabe