SeisComP3/seiscomp3 #243: Populate unit field of amplitude measurement

calum-chamberlain, 2019-09-11:
Hi all, I'm not a Seiscomp user myself, but I do make heavy usage of the GeoNet (New Zealand) catalogue, which is built using Seiscomp.  Thier catalogue is missing units for amplitude measurements (see [this issue](https://github.com/GeoNet/help/issues/60)): looking at the QuakeML docs it looks like the units for amplitudes should be SI, but those in the GeoNet catalogue are in either mm or mm/s and the `unit` field of the amplitude is not populated.  It would be *really* handy if this was populated, and the QuakeML spec encourages it.  

As an extra - can the `methodID` field also be populated with a link to the appropriate page in the seiscomp docs for various fields so that catalogue users can easily find how processing was undertaken?

jsaul, 2019-09-11:
The issue of "missing" (or in fact "implicit") amplitude units in QuakeML is something that would deserve improvement. Currently the amplitude units are the ones usually used for a certain magnitude type. Some are micrometers/second, others are millimeters, for instance. All are magnitude specific and it is assumed that the magnitude computation is well documented and homogeneous. In a real world this is not always the case, of course.

There are several possibilities to work around this. The use of methodID could be one, but what "standard" should it refer to? Another possibility would be the use of multipliers. E.g. if an amplitude is given in micrometers the multiplier would be 10E6. This would also allow to be more explicit regarding the 2080 vs. 2800 gain issue in ML for instance. But it still doesn't capture the actual unit (meters vs. meters/second or meters/second**2 etc.)

While a SeisComP specific solution can be implemented, I believe that the better place to discuss and eventually solve the issue would be the QuakeML community and/or the IASPEI magnitude working group.

gempa-jabe, 2019-09-11:
That is implemented for quite some time now, see https://github.com/SeisComP3/seiscomp3/blob/build/jakarta/2018.327/src/trunk/libs/seiscomp3/datamodel/exchange/quakeml.cpp#L857. I don't know how GeoNet creates their QuakeML files. If they use the built-in QuakeML converter (C++) then it should work as expected.

jsaul, 2019-09-11:
Sorry for the confusion, I did not have that on my radar.

calum-chamberlain, 2019-09-11:
Thanks for that! I don't know either, I will chase more on their end! Thanks for your patience.  Happy to close this.

calum-chamberlain, 2019-09-11:
Actually I might leave this open in case @salichon wants to comment.

salichon, 2019-09-11:
Great thanks  @calum-chamberlain @gempa-jabe - i ll follow up on this and see how indeed it s been implemented into the Geonet quakeml services (hence not the built in quakeml most likely).
Cheers
jerome

salichon, 2019-09-11:
I verified that our process is not converting  into SI units and unit labeeling 
Our conversion would benefit from these kind improvements ! :)

salichon, 2019-09-17:
- I think this issue can be closed ->  related to our external on-the-fly  conversion process

closed 2019-09-17 by calum-chamberlain