SeisComP3/seiscomp3 #248: Bug in FDSNWS configuration

GarryLai, 2019-09-17:
When I set the evaluation mode in FDSNWS configuration file. The FDSNWS service could not start but it didn't recorded any error in log file (with log level 4). Is this a bug?

gempa-stephan, 2019-09-18:
I can't reproduce your observation. Which FDSNWS version are you running?

I tested the FDSNWS with
*  empty evaluationMode 
* `evaluationMode = manual`
* `evaluationMode = automatic`
* `evaluationMode = foo`

The first 3 cases returned the expected result for an event query. In the last case the FDSNWS refused to start with the error message:

`invalid evaluation mode string: foo`

GarryLai, 2019-09-18:
Oh sorry, I mistook evaluation mode for evaluation status. But is there any way to restrict evaluation status in FDSNWS?

gempa-jabe, 2020-12-07:
No, you can't do that.

closed 2020-12-07 by gempa-jabe