SeisComP3/seiscomp3 #259: Fix for errors with clang - case value is not a constant expression & error invalid operands to binary expression ('__bind<

gilcel, 2020-02-28:
Hi,

I've just compiled the latest master version of seiscomp3 with clang 11 (C++11 enabled) on macOS with QT5 and there seems 3 compilation errors. Here are my fixes:

First compile error in `magnitudeview.cpp:261:10: error: case value is not a constant expression
                                case Less:`

The fix in src/gui-qt4/libs/seiscomp3/gui/datamodel/magnitudeview.cpp at line 260:
Change code from:
`			switch ( _op ) {`
to:
`			switch ( this->_op ) {`

Compilation error in iosystem.cc & arclink.cc:
`error invalid operands to binary expression ('__bind<int &, sockaddr *, unsigned long>' and 'int')`

Here's a fix:

Change in arclink.cc, line 3457 from:
`    if(bind(listenfd, (struct sockaddr *) &inet_addr, sizeof(inet_addr)) < 0)
        throw ArclinkLibraryError("bind error");`
to
`    if(::bind(listenfd, (struct sockaddr *) &inet_addr, sizeof(inet_addr)) < 0)
        throw ArclinkLibraryError("bind error");`

The same for iosystem.cc at line 2772:
Change:
`    if(bind(listenfd, (struct sockaddr *) &inet_addr, sizeof(inet_addr)) < 0)
        throw ArclinkLibraryError("bind error");`
to
`    if(::bind(listenfd, (struct sockaddr *) &inet_addr, sizeof(inet_addr)) < 0)
        throw ArclinkLibraryError("bind error");`

Then everything just compiles fine.

gempa-jabe, 2020-03-12:
According to the magnitudeview.cpp fix it seems to be a clang bug, see https://stackoverflow.com/questions/55359614/clang-complains-about-constexpr-function-in-case-for-switch-statement. Which clang version were you using?

gilcel, 2020-03-12:
OK thanks. I compiled with the latest Apple clang version 11.0.0 (clang-1100.0.33.17) which comes with Xcode 11.3.1 for macOS 10.14 and later.

gempa-jabe, 2020-03-12:
I have applied all fixes, thank you for reporting.

closed 2020-03-12 by gempa-jabe