SeisComP3/seiscomp3 #262: arclink "unexpected eof cmd" when requested via webdc3

filefolder, 2020-03-18:
noticing this error in the debug log but only when using the webdc3 client. in the web client request window it says "status: ERROR, size: 0, info: internal error"

a similar arclink [bug](https://github.com/SeisComP3/seiscomp3/issues/135) had been posted/fixed here a few years back, attributed to "python wrapper issues", could it be the same problem? there are still some instances of "Config.Exception" in the code in various spots but it's all a bit over my head.

(current dev version)

```
Wed Mar 18 15:27:44 2020 - arclink: 150.203.69.61:36210 : opening connection
Wed Mar 18 15:27:44 2020 - arclink: 150.203.69.61:36210 : HELLO
Wed Mar 18 15:27:44 2020 - arclink: 150.203.69.61:36210 : USER seis
Wed Mar 18 15:27:44 2020 - arclink: 150.203.69.61:36210 : USER_IP 150.203.9.201
Wed Mar 18 15:27:44 2020 - arclink: 150.203.69.61:36210 : LABEL WI:cffe01e2-68d0-11ea-987b-509a4c60458f:Package_1584498671250
Wed Mar 18 15:27:44 2020 - arclink: 150.203.69.61:36210 : REQUEST WAVEFORM format=MSEED
Wed Mar 18 15:27:44 2020 - arclink: 150.203.69.61:36210 : END
Wed Mar 18 15:27:44 2020 - arclink: 150.203.69.61:36210 : closing connection
Wed Mar 18 15:27:44 2020 - arclink: [30296] MESSAGE internal error
Wed Mar 18 15:27:44 2020 - arclink: [30296] ERROR
Wed Mar 18 15:27:46 2020 - arclink: [30296] unexpected eof cmd*****************
Wed Mar 18 15:27:46 2020 - arclink: [30296] terminated on signal 11
Wed Mar 18 15:27:46 2020 - arclink: [25335] starting shell
```

filefolder, 2020-03-20:
quick update, replaced all instances of Config.Exception (but NOT _Config.Exception) with just Exception, did not appear to fix anything.

andres-h, 2020-03-20:
"internal error" means an uncaught Python exception. It comes from [here](https://github.com/SeisComP3/seiscomp3/blob/48d248967bf7b4409030c5af414d9bd60c29049b/src/arclink/libs/python/seiscomp/arclink/handler.py#L1611).

The "raise" statement should make sure that the exception is re-raised and stack trace is printed to stdout. You should find it in some log file (or try to run arclink in terminal). Without stack trace, it is impossible to say what went wrong.

Since ArcLink is deprecated, please also consider using FDSNWS instead of ArcLink. It is supported by the current WebDC version.

filefolder, 2020-06-13:
i see that arclink is now missing from the new version (whew) so happy to close this

closed 2020-06-13 by filefolder