SeisComP3/seiscomp3 #266: FDSNWS dataselect: undesired mseed data packets from stations with gaps

petrrr, 2020-05-11:
We observe that we occasionally we get data back from our fdsnws dataselect service which are not anyhow relevant to the respective data request. 

After a closer analysis we have realized that this occurs in presence of data with gaps coincidence with the requested window, more precisely when the start time coincide with the a gap in the data stream. We deduce that this is due to an inappropriate assumption of data continuity which results not valid in the case of gaps. 

Here the details:

```
curl --output out1.mseed 'http://webservices.ingv.it/fdsnws/dataselect/1/query?net=*&sta=MELA,MSAG,T1299&cha=HHE,HHN,HHZ,HNE,HNN,HNZ,EHE,EHN,EHZ&starttime=2017-08-08T04:32:53.2800&endtime=2017-08-08T04:39:53.2800&nodata=404'
```

* [out1.mseed.zip](https://github.com/SeisComP3/seiscomp3/files/4610081/out1.mseed.zip)

```
IV_MSAG__HNZ      2017,220,00:01:06.600000 2017,220,00:01:08.800000 100 221
IV_MSAG__HHZ      2017,220,00:05:26.600000 2017,220,00:05:27.700000 100 111
IV_MELA__HNN      2017,220,01:58:53.310200 2017,220,01:58:55.365200 200 412
IV_MELA__HNE      2017,220,01:58:53.460200 2017,220,01:58:55.515200 200 412
IV_MELA__HHZ      2017,220,03:52:02.320200 2017,220,03:52:06.430200 100 412
IV_T1299__EHN     2017,220,03:56:02.140100 2017,220,03:56:06.250100 100 412
```
 
As can be seen there is one single data packet well before the requested time window and with no direct relation to the request, while the requested data is missing.

Looking into the archived data reveals a pattern. When the requested time window starts in a gap, the service in any case returns the preceding packages, without checking its "endtime".

* `IV_MSAG__HNZ      2017,220,00:01:06.600000 2017,220,00:01:08.800000 100 221`
```
> msi archive/2017/IV/MSAG/HNZ.D/*2017.220*
[...]
IV_MSAG__HNZ, 000001, D, 512, 221 samples, 100 Hz, 2017,220,00:01:06.600000
IV_MSAG__HNZ, 000001, D, 512, 221 samples, 100 Hz, 2017,220,06:24:06.600000
[...]
```

* `IV_MSAG__HHZ      2017,220,00:05:26.600000 2017,220,00:05:27.700000 100 111`
```
> msi archive/2017/IV/MSAG/HHZ.D/*2017.220*
[...]
IV_MSAG__HHZ, 000001, D, 512, 111 samples, 100 Hz, 2017,220,00:05:26.600000
IV_MSAG__HHZ, 000001, D, 512, 112 samples, 100 Hz, 2017,220,06:21:57.710000
[...]
```

* `IV_MELA__HNN      2017,220,01:58:53.310200 2017,220,01:58:55.365200 200 412`
```
> msi archive/2017/IV/MELA/HNN.D/*2017.220*
[...]
IV_MELA__HNN, 000001, D, 512, 412 samples, 200 Hz, 2017,220,01:58:53.310200
IV_MELA__HNN, 000001, D, 512, 412 samples, 200 Hz, 2017,220,07:17:23.400200
[...]
```

* `IV_MELA__HNE      2017,220,01:58:53.460200 2017,220,01:58:55.515200 200 412`
```
> msi archive/2017/IV/MELA/HNE.D/*2017.220*
[...]
IV_MELA__HNE, 000001, D, 512, 412 samples, 200 Hz, 2017,220,01:58:53.460200
IV_MELA__HNE, 000001, D, 512, 412 samples, 200 Hz, 2017,220,07:55:10.710200
[...]
```

* `IV_MELA__HHZ      2017,220,03:52:02.320200 2017,220,03:52:06.430200 100 412`
```
> msi archive/2017/IV/MELA/HHZ.D/*2017.220*
[...]
IV_MELA__HHZ, 000001, D, 512, 412 samples, 100 Hz, 2017,220,03:52:02.320200
IV_MELA__HHZ, 000001, D, 512, 400 samples, 100 Hz, 2017,220,10:58:34.600200
[...]
```

* `IV_T1299__EHN     2017,220,03:56:02.140100 2017,220,03:56:06.250100 100 412`
```
msi archive/2017/IV/T1299/EHN.D/*2017.220*
[...]
IV_T1299__EHN, 000001, D, 512, 412 samples, 100 Hz, 2017,220,03:56:02.140100
IV_T1299__EHN, 000001, D, 512, 412 samples, 100 Hz, 2017,220,04:38:43.300100
IV_T1299__EHN, 000001, D, 512, 412 samples, 100 Hz, 2017,220,05:21:19.360300
[...]
```

Apparently the procedure extracting the data simple searches for the last data packet which precedes the requested start time (t_stamp1 < t_start_req ). It serves data from this data packet until the data packet with timestamp large end time (t_stamp2 > t_end_req). For continues data this is might be appropriate. 

However, the more general approach would be to check if the requested start time falls into the time span of the found data packet  ( tstamp1=t_dpacket1_start < t_start_req <= t_dpacket1_end).

andres-h, 2020-05-11:
Please make sure:

sysop@st27dmz:~$ grep service seiscomp3/etc/fdsnws.cfg
#recordstream.service = fastsds
recordstream.service = sdsarchive

petrrr, 2020-05-11:
Here it is:

```
sysop@eidaws:~/seiscomp3/etc$ grep service ~/seiscomp3/etc/fdsnws.cfg
# services to limit main memory consumption, default: 10000
# Enables/disables the Event service.
# Enables/disables the Station service. 
# Enables/disables the DataSelect service.
recordstream.service = sdsarchive
#recordstream.service = sdsarchive
```

```
sysop@eidaws:~/seiscomp3/etc$ grep service ~/seiscomp3/etc/defaults/fdsnws.cfg
# Name of the recordstream service implementation.
recordstream.service = sdsarchive
```

So I'd say `sdsarchive` is active not `fastsds` as you would expect.

gempa-jabe, 2020-05-11:
Which version of SeisComP are you running and what is your operating system?

gempa-stephan, 2020-05-11:
Also please use ```scdumpcfg fdsnws``` to check which configuration variables are in place since every SeisComP application reads from many configuration files. In your case the ```~/seiscomp3/etc/fdsnws.cfg``` might be overruled by variables set in ```~/.seiscomp3/fdsnws.cfg```.

gempa-jabe, 2020-05-12:
I have tested such a case with the latest SeisComP3 version and it works as expected.

petrrr, 2020-05-12:
Hi @gempa-stephan!

There seems to be no config file `~/.seiscomp3/fdsnws.cfg` in place.

``` 
sysop@eidaws:~$ ls -la ~/.seiscomp3/*.cfg
ls: cannot access '/home/sysop/.seiscomp3/*.cfg': No such file or directory
```

```
sysop@eidaws:~$ scdumpcfg fdsnws
core.plugins
  value(s) : dbmysql
  source   : /home/sysop/seiscomp3/etc/global.cfg
logging.level
  value(s) : 4
  source   : /home/sysop/seiscomp3/etc/fdsnws.cfg
logging.file.rotator
  value(s) : true
  source   : /home/sysop/seiscomp3/etc/defaults/global.cfg
logging.file.rotator.timeSpan
  value(s) : 86400
  source   : /home/sysop/seiscomp3/etc/defaults/global.cfg
logging.file.rotator.archiveSize
  value(s) : 7
  source   : /home/sysop/seiscomp3/etc/defaults/global.cfg
connection.server
  value(s) : localhost
  source   : /home/sysop/seiscomp3/etc/defaults/global.cfg
connection.timeout
  value(s) : 3
  source   : /home/sysop/seiscomp3/etc/defaults/global.cfg
connection.encoding
  value(s) : binary
  source   : /home/sysop/seiscomp3/etc/defaults/global.cfg
recordstream.service
  value(s) : sdsarchive
  source   : /home/sysop/seiscomp3/etc/fdsnws.cfg
recordstream.source
  value(s) : /mnt/archive_aufs
  source   : /home/sysop/seiscomp3/etc/fdsnws.cfg
agencyID
  value(s) : INGV
  source   : /home/sysop/seiscomp3/etc/global.cfg
organization
  value(s) : Italian Seismic Data Center
  source   : /home/sysop/seiscomp3/etc/global.cfg
picker.filters
  value(s) : BP 0.1 - 1 Hz  3rd order;BW(3,0.1,1), BP 0.1 - 2 Hz  3rd order;BW(3,0.1,2), BP 0.4 - 1 Hz  3rd order;BW(3,0.4,1), @BP 0.7 - 2 Hz  3rd order;BW(3,0.7,2), BP 1 - 3 Hz  3rd order;BW(3,1.0,3), BP 2 - 4 Hz  3rd order;BW(3,2.0,4), BP 3 - 6 Hz  3rd order;BW(3,3.0,6), BP 4 - 8 Hz  3rd order;BW(3,4.0,8), BP 1 - 5 Hz  3rd order;BW(3,1.0,5), BP 0.7 - 2 Hz + STA/LTA(1,50);RMHP(10)->ITAPER(30)->BW(3,0.7,2)->STALTA(1,50)
  source   : /home/sysop/seiscomp3/etc/defaults/global.cfg
eventlist.visibleColumns
  value(s) : M, TP, Phases, Lat, Lon, Depth, Stat, Agency, Region, ID
  source   : /home/sysop/seiscomp3/etc/defaults/global.cfg
ttt.libtau.tables
  value(s) : iasp91, ak135
  source   : /home/sysop/seiscomp3/etc/defaults/global.cfg
ttt.LOCSAT.tables
  value(s) : iasp91, tab
  source   : /home/sysop/seiscomp3/etc/defaults/global.cfg
plugins
  value(s) : fdsnxml, fdsnxml
  source   : /home/sysop/seiscomp3/etc/fdsnws.cfg
recordBulkSize
  value(s) : 102400
  source   : /home/sysop/seiscomp3/etc/defaults/fdsnws.cfg
datacenterID
  value(s) : INGV
  source   : /home/sysop/seiscomp3/etc/global.cfg
listenAddress
  value(s) : 0.0.0.0
  source   : /home/sysop/seiscomp3/etc/fdsnws.cfg
port
  value(s) : 8080
  source   : /home/sysop/seiscomp3/etc/fdsnws.cfg
queryObjects
  value(s) : 250000
  source   : /home/sysop/seiscomp3/etc/fdsnws.cfg
samplesM
  value(s) : 1000
  source   : /home/sysop/seiscomp3/etc/fdsnws.cfg
connections
  value(s) : 5
  source   : /home/sysop/seiscomp3/etc/fdsnws.cfg
accessLog
  value(s) : @ROOTDIR@/var/log/fdsnws_accessLog.log
  source   : /home/sysop/seiscomp3/etc/fdsnws.cfg
serveEvent
  value(s) : false
  source   : /home/sysop/seiscomp3/etc/fdsnws.cfg
serveStation
  value(s) : true
  source   : /home/sysop/seiscomp3/etc/fdsnws.cfg
serveDataSelect
  value(s) : true
  source   : /home/sysop/seiscomp3/etc/fdsnws.cfg
trackdb.enable
  value(s) : true
  source   : /home/sysop/seiscomp3/etc/fdsnws.cfg
trackdb.defaultUser
  value(s) : fdsnws@ingv
  source   : /home/sysop/seiscomp3/etc/fdsnws.cfg
database.type
  value(s) : mysql
  source   : /home/sysop/seiscomp3/etc/fdsnws.cfg
database.parameters
  value(s) : sysop:sysop@localhost/seiscomp3
  source   : /home/sysop/seiscomp3/etc/fdsnws.cfg
auth.enable
  value(s) : true
  source   : /home/sysop/seiscomp3/etc/fdsnws.cfg
useArclinkAccess
  value(s) : true
  source   : /home/sysop/seiscomp3/etc/fdsnws.cfg
```

gempa-jabe, 2020-05-12:
I have requested a day long file for IV.MSAG..HNZ and I think I could reproduce the issue. This data is indeed interesting ...

petrrr, 2020-05-12:
@gempa-jabe: 
According to our record it seems we are still running release 2018.327 on a Debian 9 system. It seems to have been installed in February 2019. I understand there is no newer release, but maybe new packages are available (patch level 19 ???).

However, I do not know how to verify the which package or version exactly is currently deployed.

petrrr, 2020-05-12:
In case you need further data directly from the archive for testing, just let me know.

gempa-jabe, 2020-05-12:
```
fdsnws --version
```

petrrr, 2020-05-12:
@gempa-jabe:
```
sysop@eidaws:~$ fdsnws --version
fdsnws: Jakarta 2018.327.p6
API version: 12.0.0
GIT HEAD: 
Compiler: c++ (Debian 6.3.0-18) 6.3.0 20170516
Build system: Linux 4.9.0-3-amd64
OS: Debian GNU/Linux 9.1 (stretch) / Linux
```

gempa-jabe, 2020-05-12:
I have fixed it. It was indeed a tricky issue which just shows if a gap follows the first physical record and that is before the requested start time.

petrrr, 2020-05-12:
@gempa-jabe: Will you provide the fix or are you planing to provide new packages?

gempa-jabe, 2020-05-12:
I will bundle a new patch release in a couple of minutes. New version will be 2018.327p21.

closed 2020-12-07 by gempa-jabe