SeisComP3/seiscomp3 #270: FilterRule instance has no attribute 'Restricted'

jbienkowski, 2020-06-30:
Hi all,

Today after a routine SeisComP3 restart we have noticed that `fdsnws` module is not starting up properly.
In the logs we have found following lines:
```
2020/06/30 08:19:50 [notice/log] reading inventory filter file: /home/orfeus/seiscomp3/etc/stationFilter.ini
2020/06/30 08:28:16 [error/log] FilterRule instance has no attribute 'Restricted'
2020/06/30 08:28:16 [info/MYSQL] Disconnecting from database
2020/06/30 08:28:16 [notice/Application] Shutdown
```
After checking the integrity of our metadata inventory and stationFilter.ini file, we found that there might be something wrong in `fdsnws.py` lines 1318 - 1329 (https://github.com/SeisComP3/seiscomp3/blame/master/src/trunk/apps/fdsnws/fdsnws.py#L1318):
```
if rule.restricted is not None:
    try:
        if cha.restricted() != rule.restricted:
            continue
    except ValueError:
        if staRestricted is not None:
            if sta.Restricted != rule.Restricted:
                continue
        elif netRestricted is None or \
                netRestricted != rule.Restricted:
            continue
```

We have changed it to:
```
# restricted
if rule.restricted is not None:
    try:
        if cha.restricted() != rule.restricted:
            continue
    except ValueError:
        if staRestricted is not None:
            if sta.restricted() != rule.restricted:
                continue
        elif netRestricted is None or \
                netRestricted != rule.restricted:
            continue
```
Now everything seems to be working fine, but I would appreciate if you could review this and patch the master branch if needed.

Thanks,
Jarek Bienkowski

gempa-stephan, 2020-07-01:
The code is indeed flawed and there a more wrong variables when comparing the shared status. I'll create a fix. Meanwhile replace it as follows:

```
                            # restricted
                            if rule.restricted is not None:
                                try:
                                    if cha.restricted() != rule.restricted:
                                        continue
                                except ValueError:
                                    if staRestricted is not None:
                                         if staRestricted != rule.restricted:
                                            continue
                                    elif netRestricted is None or \
                                            netRestricted != rule.restricted:
                                        continue

                            # shared
                            if rule.shared is not None:
                                try:
                                    if cha.shared() != rule.shared:
                                        continue
                                except ValueError:
                                    if staShared is not None:
                                        if staShared != rule.shared:
                                            continue
                                    elif netShared is None or \
                                            netShared != rule.shared:
                                        continue
```

gempa-stephan, 2020-07-01:
The fix was implemented in 81c814a0. Also I added a test case to cover the filter feature.

closed 2020-07-01 by gempa-stephan