SeisComP3/seiscomp3 #271: Compilation issues due to file advancedoriginsymbols.cpp

a-oth, 2020-09-22:
Hello,
Compiling seiscomp3 on macOS, I run into the following issues:
```
/Users/adrien/Computing/seiscomp3-git/seiscomp3-macos-jakarta-devel-2020-09-22/src/gui-qt4/libs/seiscomp3/gui/datamodel/advancedoriginsymbol.h:82:31: error: 
      field has incomplete type 'QPainterPath'
                QPainterPath                _renderPath;
                                            ^
/usr/local/opt/qt/lib/QtGui.framework/Headers/qmatrix.h:54:7: note: forward declaration of 'QPainterPath'
class QPainterPath;
      ^
/Users/adrien/Computing/seiscomp3-git/seiscomp3-macos-jakarta-devel-2020-09-22/src/gui-qt4/libs/seiscomp3/gui/datamodel/advancedoriginsymbol.cpp:87:2: error: 
      no matching member function for call to 'setLocation'
        setLocation(0.0, 0.0);
        ^~~~~~~~~~~
/Users/adrien/Computing/seiscomp3-git/seiscomp3-macos-jakarta-devel-2020-09-22/src/gui-qt4/libs/seiscomp3/gui/map/mapsymbol.h:125:8: note: 
      candidate function not viable: no known conversion from 'Seiscomp::Gui::AdvancedOriginSymbol' to 'Seiscomp::Gui::Map::Symbol' for
      object argument
                void setLocation(double latitude, double longitude);
                     ^
/Users/adrien/Computing/seiscomp3-git/seiscomp3-macos-jakarta-devel-2020-09-22/src/gui-qt4/libs/seiscomp3/gui/map/mapsymbol.h:126:8: note: 
      candidate function not viable: requires single argument 'latlon', but 2 arguments were provided
                void setLocation(const QPointF &latlon);
                     ^
/Users/adrien/Computing/seiscomp3-git/seiscomp3-macos-jakarta-devel-2020-09-22/src/gui-qt4/libs/seiscomp3/gui/datamodel/advancedoriginsymbol.cpp:88:2: error: 
      cannot initialize object parameter of type 'Seiscomp::Gui::OriginSymbol' with an expression of type
      'Seiscomp::Gui::AdvancedOriginSymbol'
        setDepth(0.0);
        ^~~~~~~~
/Users/adrien/Computing/seiscomp3-git/seiscomp3-macos-jakarta-devel-2020-09-22/src/gui-qt4/libs/seiscomp3/gui/datamodel/advancedoriginsymbol.cpp:89:2: error: 
      cannot initialize object parameter of type 'Seiscomp::Gui::OriginSymbol' with an expression of type
      'Seiscomp::Gui::AdvancedOriginSymbol'
        setPreferredMagnitudeValue(0.0);
        ^~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/adrien/Computing/seiscomp3-git/seiscomp3-macos-jakarta-devel-2020-09-22/src/gui-qt4/libs/seiscomp3/gui/datamodel/advancedoriginsymbol.cpp:97:2: error: 
      no matching member function for call to 'setLocation'
        setLocation(_origin->latitude(), _origin->longitude());
        ^~~~~~~~~~~
/Users/adrien/Computing/seiscomp3-git/seiscomp3-macos-jakarta-devel-2020-09-22/src/gui-qt4/libs/seiscomp3/gui/map/mapsymbol.h:125:8: note: 
      candidate function not viable: no known conversion from 'Seiscomp::Gui::AdvancedOriginSymbol' to 'Seiscomp::Gui::Map::Symbol' for
      object argument
                void setLocation(double latitude, double longitude);
                     ^
/Users/adrien/Computing/seiscomp3-git/seiscomp3-macos-jakarta-devel-2020-09-22/src/gui-qt4/libs/seiscomp3/gui/map/mapsymbol.h:126:8: note: 
      candidate function not viable: requires single argument 'latlon', but 2 arguments were provided
                void setLocation(const QPointF &latlon);
                     ^
/Users/adrien/Computing/seiscomp3-git/seiscomp3-macos-jakarta-devel-2020-09-22/src/gui-qt4/libs/seiscomp3/gui/datamodel/advancedoriginsymbol.cpp:100:3: error: 
      cannot initialize object parameter of type 'Seiscomp::Gui::OriginSymbol' with an expression of type
      'Seiscomp::Gui::AdvancedOriginSymbol'
                setDepth(depth);
                ^~~~~~~~
/Users/adrien/Computing/seiscomp3-git/seiscomp3-macos-jakarta-devel-2020-09-22/src/gui-qt4/libs/seiscomp3/gui/datamodel/advancedoriginsymbol.cpp:106:3: error: 
      cannot initialize object parameter of type 'Seiscomp::Gui::OriginSymbol' with an expression of type
      'Seiscomp::Gui::AdvancedOriginSymbol'
                setPreferredMagnitudeValue(magnitude);
                ^~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/adrien/Computing/seiscomp3-git/seiscomp3-macos-jakarta-devel-2020-09-22/src/gui-qt4/libs/seiscomp3/gui/datamodel/advancedoriginsymbol.cpp:174:2: error: 
      use of undeclared identifier 'sincos'; did you mean '__sincos'?
        sincos(deg2rad(90 - azimuth), &sinAzi, &cosAzi);
        ^~~~~~
        __sincos
/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/math.h:655:29: note: '__sincos' declared here
__header_always_inline void __sincos(double __x, double *__sinp, double *__cosp) {
                            ^
/Users/adrien/Computing/seiscomp3-git/seiscomp3-macos-jakarta-devel-2020-09-22/src/gui-qt4/libs/seiscomp3/gui/datamodel/advancedoriginsymbol.cpp:175:17: error: 
      cannot initialize object parameter of type 'const Seiscomp::Gui::Map::Symbol' with an expression of type
      'Seiscomp::Gui::AdvancedOriginSymbol'
        sincos(deg2rad(latitude()), &sinLat, &cosLat);
                       ^~~~~~~~
/Users/adrien/Computing/seiscomp3-git/seiscomp3-macos-jakarta-devel-2020-09-22/src/gui-qt4/../trunk/libs/seiscomp3/math/math.h:24:27: note: 
      expanded from macro 'deg2rad'
#define deg2rad(d) (M_PI*(d)/180.0)
                          ^
/Users/adrien/Computing/seiscomp3-git/seiscomp3-macos-jakarta-devel-2020-09-22/src/gui-qt4/libs/seiscomp3/gui/datamodel/advancedoriginsymbol.cpp:188:3: error: 
      use of undeclared identifier 'sincos'; did you mean '__sincos'?
                sincos(theta, &y, &x);
                ^~~~~~
                __sincos
/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/math.h:655:29: note: '__sincos' declared here
__header_always_inline void __sincos(double __x, double *__sinp, double *__cosp) {
                            ^
/Users/adrien/Computing/seiscomp3-git/seiscomp3-macos-jakarta-devel-2020-09-22/src/gui-qt4/libs/seiscomp3/gui/datamodel/advancedoriginsymbol.cpp:200:3: error: 
      use of undeclared identifier 'sincos'; did you mean '__sincos'?
                sincos(delta, &sinDelta, &cosDelta);
                ^~~~~~
                __sincos
/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/math.h:655:29: note: '__sincos' declared here
__header_always_inline void __sincos(double __x, double *__sinp, double *__cosp) {
                            ^
/Users/adrien/Computing/seiscomp3-git/seiscomp3-macos-jakarta-devel-2020-09-22/src/gui-qt4/libs/seiscomp3/gui/datamodel/advancedoriginsymbol.cpp:208:9: error: 
      cannot initialize object parameter of type 'const Seiscomp::Gui::Map::Symbol' with an expression of type
      'Seiscomp::Gui::AdvancedOriginSymbol'
                if ( (latitude() - 90.0) > 1.0e-8 )
                      ^~~~~~~~
/Users/adrien/Computing/seiscomp3-git/seiscomp3-macos-jakarta-devel-2020-09-22/src/gui-qt4/libs/seiscomp3/gui/datamodel/advancedoriginsymbol.cpp:210:13: error: 
      cannot initialize object parameter of type 'const Seiscomp::Gui::Map::Symbol' with an expression of type
      'Seiscomp::Gui::AdvancedOriginSymbol'
                else if ((latitude() + 90.0) < 1.0e-8)
                          ^~~~~~~~
/Users/adrien/Computing/seiscomp3-git/seiscomp3-macos-jakarta-devel-2020-09-22/src/gui-qt4/libs/seiscomp3/gui/datamodel/advancedoriginsymbol.cpp:216:13: error: 
      cannot initialize object parameter of type 'const Seiscomp::Gui::Map::Symbol' with an expression of type
      'Seiscomp::Gui::AdvancedOriginSymbol'
                it->lon = longitude();
                          ^~~~~~~~~
/Users/adrien/Computing/seiscomp3-git/seiscomp3-macos-jakarta-devel-2020-09-22/src/gui-qt4/libs/seiscomp3/gui/datamodel/advancedoriginsymbol.cpp:288:16: error: 
      cannot initialize object parameter of type 'Seiscomp::Gui::Map::Symbol' with an expression of type
      'Seiscomp::Gui::AdvancedOriginSymbol'
        OriginSymbol::calculateMapPosition(canvas);
                      ^~~~~~~~~~~~~~~~~~~~
/Users/adrien/Computing/seiscomp3-git/seiscomp3-macos-jakarta-devel-2020-09-22/src/gui-qt4/libs/seiscomp3/gui/datamodel/advancedoriginsymbol.cpp:314:16: error: 
      cannot initialize object parameter of type 'Seiscomp::Gui::OriginSymbol' with an expression of type
      'Seiscomp::Gui::AdvancedOriginSymbol'
        OriginSymbol::customDraw(canvas, painter);
                      ^~~~~~~~~~
16 errors generated.
```

I did also encounter the `use of undeclared identifier 'sincos'; did you mean '__sincos'?`errors when compiling seiscomp4 (and could solve it by simply adding the underscores), but none of the other issues.

Is this a macOS specific problem?

Many thanks
Adrien

gempa-stephan, 2020-09-22:
Just add `#include <QPainterPath>` add the top of `src/gui-qt4/libs/seiscomp3/gui/datamodel/advancedoriginsymbol.h`. I'll push an update soon.

Regarding `sincos`, try adding `#include <math.h>` to the includes of `src/gui-qt4/libs/seiscomp3/gui/datamodel/advancedoriginsymbol.cpp`. If this change has no effect we may add an Apple specific ifdef, e.g.

```
#ifdef __APPLE__
#define sincos(x, s, c) __sincos(x, s, c)
#endif
```

gempa-jabe, 2020-09-22:
I have pushed a change, please test again.

a-oth, 2020-09-22:
Thanks, Stephan and Jan, I have just tested, the only errors that I get are now the sincos ones still.

gempa-jabe, 2020-09-22:
Then we go with the approach proposed by @gempa-stephan.

a-oth, 2020-09-22:
Sounds good to me, thanks!

gempa-jabe, 2020-09-22:
Please give it another try.

a-oth, 2020-09-22:
I'll try this afternoon, but I might not get around it since I am out of office - at the latest tomorrow morning, I will let you know! Thanks!

a-oth, 2020-09-23:
Hi Jan, Stephan,
I tried to compile again, and receive the same error. If I replace `#ifdef MACOSX`  with `#ifdef __APPLE__`, then it compiles.

Cheers
Adrien

gempa-jabe, 2020-09-23:
I added the correct header to provide the MACOSX definition. Please try again.

a-oth, 2020-09-23:
With this change it compiles fine, thanks!

closed 2020-09-25 by gempa-jabe