SeisComP3/seiscomp3 #272: FDSNWS event blacklist nor working

KDFischer, 2020-12-01:
Hi,
I just upgraded to seiscomp3-jakarta-2020.330 and noticed that the fdsnws does not handles the configuration parameter `eventType.blacklist` correctly. I used to have `eventType.blacklist = "outside of network interest", "not existing", "duplicate", "not locatable"` which worked fine in older versions. Now a query lists me only blacklisted events.

To pin-point this unexpected behavior I tried `eventType.blacklist = "foobar"`. This resulted in the error `error parsing eventType.whitelist: event type name 'foobar' not supported`in the log file. Looking at the code I think that this is just a type in the error message. The real cause might be somewhere in FDSNEvent(). Therefor the _old_ fdsnws script didn't work either.

I didn't use eventType.whitelist in my configuartion, i. e. I want all events except the blacklisted types.

I hope you can fix this.

Cheers,

Kasper

gempa-stephan, 2020-12-02:
Hi Kasper,

Thanks for reporting this issue. I identified the cause and committed a fix in 2025af2. We will release 2020.330.p1 soon. Meanwhile you may just exchange your `seiscomp3/bin/fdsnws` with https://github.com/SeisComP3/seiscomp3/blob/master/src/trunk/apps/fdsnws/fdsnws.py.

Cheers,
Stephan

KDFischer, 2020-12-02:
Hi Stephan,
thanks for the quick fix. The updated `fdsnws.py` fixed the problem for me.
Cheers,
Kasper

closed 2020-12-02 by KDFischer