SeisComP3/seiscomp3 #273: SeedLink fails to fetch data from some servers after restart

KDFischer, 2020-12-05:
Dear Seiscomp-Team,

the seedlink server coming with version seiscomp3-jakarta-2020.330 (I downloaded it from Gempa) fails to fetch data after running `seiscomp restart seedlink` from some servers / stations. I'm using the chain plugin to fetch the data from my stations or other 3rd-party servers. The failing servers run SeedLink v3.0  (2005.212). Servers running more recent versions didn't have this problem.
To reproduce this problem I set up a minimal seiscomp3 environment in a virtual machine. There I configured 4 stations (GE.IBBN, GR.BUG, GR.KAST and GR.LUEB). The station GE.IBBN uses a seedlink binding with the chain plugin and the default server `geofon.gfz-potsdam.de`. The other 3 stations are getting data from 'www.szgrf.bgr.de`. After running `seiscomp restart seedlink` no data is retrieved for the GR stations. I managed to re-enable fetching the data by running
 ```
seiscomp stop seedlink
rm ~/seiscomp3/var/run/seedlink/www.szgrf.brg.de:18000.seq
seiscomp start seedlink
```
This will create a data gap for the GR stations. Attached are two log files for the two cases: 
[seedlink_restart.log](https://github.com/SeisComP3/seiscomp3/files/5647721/seedlink_restart.log)
[seedlink_rm-seq.log](https://github.com/SeisComP3/seiscomp3/files/5647722/seedlink_rm-seq.log)

Cheers,

Kasper

KDFischer, 2020-12-06:
I tried to set `sources.chain.batchmode = false` to solve the problem, but the `seedlink.log` still reports using `BATCH` should I open a new issue for this?

gempa-jabe, 2020-12-07:
I have telneted to www.szgrf.bgr.de:18000 and the "info streams" command returns an error. Not sure if the installation works correctly.

gempa-jabe, 2020-12-07:
I found the issue and fixed it. Actually a fix was missing after upgrading libslink. This fix is already in SeisComP >= 4. I committed already to the master branch.

KDFischer, 2020-12-07:
Hi Jan,

I am not managing the server www.szgrf.bgr<http://www.szgrf.bgr>.de. Maybe you can contact Klaus Stammler (klaus.stammler@bgr.de<mailto:klaus.stammler@bgr.de>) or Mathias Hoffmann (Mathias.Hoffmann@bgr.de<mailto:Mathias.Hoffmann@bgr.de>).

Cheers,

Kasper

Dr. Kasper D. Fischer
Seismologisches Observatorium
Ruhr-Universität Bochum
IA 4/93
44780 Bochum

Tel. 0234 32-27574
kasper.fischer@ruhr-uni-bochum.de<mailto:kasper.fischer@ruhr-uni-bochum.de>

KDFischer, 2020-12-07:
Thanks for fixing this. I guess I have to compile seiscomp3 from source to test it?

gempa-jabe, 2020-12-07:
You can also use a precompiled package from https://data.gempa.de/packages/Public/seiscomp3/jakarta-2020.330/.

KDFischer, 2020-12-07:
Thank you! This resolved the issue in my test environment. 😄 👍

closed 2020-12-07 by KDFischer