SeisComP3/seiscomp3 #277: db/init/sqlite3 AUTOINCREMENT integer issue

FMassin, 2021-01-26:
Hello,

I am using:
```bash
$ sqlite3 --version
3.11.0 2016-02-15 17:29:24 3d862f207e3adc00f78066799ac5a8c282430a5f
```
and I had the following problem while initiating db with sqlite:
```bash
$ sqlite3 -batch -init $SEISCOMP_ROOT/share/db/sqlite3.sql  test_db_no_event.sqlite .exit                                           
Error: near line 73: AUTOINCREMENT is only allowed on an INTEGER PRIMARY KEY
Error: near line 91: no such table: Object
Error: near line 92: no such table: Object
Error: near line 93: no such table: Object
Error: near line 94: no such table: Object
Error: near line 95: no such table: Object
Error: near line 96: no such table: Object
Error: near line 97: no such table: Object
Error: near line 98: no such table: Object
Error: near line 99: no such table: Object
Error: near line 100: no such table: Object
Error: near line 101: no such table: Object
Error: near line 102: no such table: Object
Error: near line 103: no such table: Object
Error: near line 104: no such table: Object
Error: near line 105: no such table: Object
Error: near line 106: no such table: Object
$ scdb --plugins dbsqlite3 -d "sqlite3://test_db_no_event.sqlite" -i inventory.xml --debug
...16:12:56 [error] sqlite3 execute: no such table: Object
16:12:56 [error] writing object with publicID 'smi:ch.ethz.sed/Sensor#20150717135436.122766.10' failed
16:12:56 [error] sqlite3 execute: no such table: Object
16:12:56 [error] writing object with publicID 'smi:ch.ethz.sed/Sensor#20150717135455.166941.10' failed
16:12:56 [error] sqlite3 execute: no such table: Object
16:12:56 [error] writing object with publicID 'smi:ch.ethz.sed/Sensor/le3d5s.20181030165122.62505' failed
16:12:56 [error] sqlite3 execute: no such table: Object
```
And consequently `scxmldump -I` would not return any metadata. So I followed advice found online and edited $SEISCOMP_ROOT/share/db/sqlite3.sql   line 73:
```diff
-         _oid BIGINT PRIMARY KEY AUTOINCREMENT NOT NULL,
+         _oid INTEGER PRIMARY KEY AUTOINCREMENT,
```

This version of the sqlite3 init script can then be used without triggering errors during neither of the above steps.

But am I correct? 

Cheers

Fred

gempa-jabe, 2021-02-10:
Yes, you are correct. AUTOINCREMENT is only available for INTEGER in SQLite3. We will fix that.

gempa-jabe, 2021-03-08:
Fixed in release 2020.330.

closed 2021-03-08 by gempa-jabe