SeisComP3/seiscomp3 #278: problem with fdsnws dataselect

acpardo, 2021-03-04:
Hello,
I have a problem when requesting miniseed data with dataselect in the following context:

* metadata exists for a period (for exemple 2020-01-01 to present)
* for this period there are miniseed files for some days (not contiguous and in the middle of the period) in the archive.
* if I request miniseed data for exemple from 2020-01-01 to 2020-01-31, the request ended on "nodata". To get the data archived in this period I have to put the date of the first data and end of the last data exactly. 

I have this problem in version **SC3 2020.330.04.**
I run a SC3 version **2017.334.05** in another server and it works this error does not occur.

Thanks for your help
Constanza

gempa-jabe, 2021-03-08:
Thank you for reporting. We are going to check it.

gempa-jabe, 2021-03-09:
Btw, which recordstream do you have configured? `sdsarchive://`?

gempa-jabe, 2021-03-09:
We found the issue and fixed it.

closed 2021-03-09 by gempa-jabe