SeisComP3/seiscomp3 #282: Compile on Debian 11

marcelobianchi, 2021-08-21:
Was trying to compile the sc3 on Debian 11 w/ gcc 10 + boost 1.74 and needed to change the file:

src/gui-qt4/libs/seiscomp3/gui/map/layers/geofeaturelayer.cpp

Adding the line:

#include <boost/bind.hpp>

I not so sure but i guess it is missing somehow to be able to use the _1 feature. Does not make any sense to me to add a merge request for that, specially that I don't have a clue about it so I am leaving an issue here.

-Mb

gempa-jabe, 2021-08-23:
Thank you Marcelo for letting us know. Does just replacing `_1` with `boost::placeholders::_1` work without including the header?

marcelobianchi, 2021-08-23:
Yes, it does work. 

[image]

Thank  you!

gempa-jabe, 2021-08-23:
I pushed the fix. Btw, SeisComP (>= 4) includes it already ;)

closed 2021-08-23 by gempa-jabe